RobotLocomotion/drake #13541: mac/focal yaml serialization differs from bionic's

ggould-tri, 2020-06-11:
For whatever reason, yaml serialization on mac and focal produces vertically formatted lists, while serialization on bionic produces horizontally formatted ones.  This produces errors like

```AssertionError: 'exam[21 chars]rams:\n  - 5.0\n  - 50.0\n  - 5.0\n  - 1000.0\[89 chars]05\n' != 'exam[21 chars]rams: [5.0, 50.0, 5.0, 1000.0]\n  initial_stat[59 chars]05\n'
  example:
+   controller_params: [5.0, 50.0, 5.0, 1000.0]
+   initial_state: [1.2, 0.0, 0.0, 0.0]
-   controller_params:
-   - 5.0
-   - 50.0
-   - 5.0
-   - 1000.0
-   initial_state:
-   - 1.2
-   - 0.0
-   - 0.0
-   - 0.0
    t_final: 30.0
    tape_period: 0.05

The relevant test was nerfed in #13538 but should be rewritten to compare the parsed (or post-parse serialized) rather than a raw reference serialized form in its roundtrip tests (ie, to insist on semantic rather than syntactic equality).

jwnimmer-tri, 2020-06-11:
Ideally, the I/O code could be set to use the flow style during serialization, because it's nicer to read.  We do that for C++ already:
https://github.com/RobotLocomotion/drake/blob/e55b6dd0bdd0bbb8a45a5ba9390407aaa90f2b8e/common/yaml/yaml_write_archive.cc#L29-L37

ggould-tri, 2020-06-11:
Amen on the nicer-to-read.
I believe we may have ended up on opposite sides of https://github.com/yaml/pyyaml/pull/256 and so must always set `default_flow_style` when we call the dumper.

jwnimmer-tri, 2020-06-11:
Ooh, a tribool found in the wild.  Get out the documentary cameras.

PR #13543 (2020-06-15): Force older and newer yaml.dump to give the same output
* pyyaml changed its default flow semantics in https://github.com/yaml/pyyaml/pull/256
* We must override the default with the magic tribool value `None`
* Fixes #13541

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13543)
commit 43a28f99 (2020-06-15): Force older and newer yaml.dump to give the same output (#13543)

* Force older and newer yaml.dump to give the same output

* pyyaml changed its default flow semantics in https://github.com/yaml/pyyaml/pull/256
* We must override the default with the magic tribool value `None`
* Fixes #13541
closed 2020-06-15 by ggould-tri