RobotLocomotion/drake #13548: py manipulation: Move `differential_ik` system from `drake/examples` to `pydrake.manipulation`?

EricCousineau-TRI, 2020-06-11:
See Slack / StackOverflow:
https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1591904734050100
https://stackoverflow.com/questions/62332244/access-differentialik-outside-of-drake

\cc @mpetersen94 @pvarin

EricCousineau-TRI, 2020-06-11:
@mpetersen94 I won't have time in the next few days to do this. Is it something you're OK with waiting for, or might you have time to try moving this?

jwnimmer-tri, 2020-06-12:
The code was added in #9870 by \CC @RussTedrake with a `TODO(russt): Clean this up and move it to C++.`  I am not sure that only changing its module is the correct path forward, especially with Russ' recent mention of porting the manipulation station code to C++.  Maybe a better path forward is moving it to C++ while cleaning it up, testing it, and binding it for pydrake?

EricCousineau-TRI, 2020-06-12:
That's one way to do it, and that's the more ideal. However, if I were to do it, I'd definitely only want to do the minimum amt if possible, esp. given that the request is for the Python interface.

Will mull it over some more, though. Thanks!

jwnimmer-tri, 2020-06-12:
If the need is just to call the code, then _forking_ it the minimum effort possible; it takes like 10 seconds.  To move it into a public library, the PR is anyway going to have to do the "cleanup" (fix the mutable-member-context problem, write tests, decide which methods are public, add API docs).

EricCousineau-TRI, 2020-06-12:
SGTM. @mpetersen94 You cool with forking for now?

mpetersen94, 2020-06-12:
Yes, I can just copy-pasta for now.

RussTedrake, 2020-07-05:
This is a duplicate of #11581.
It's on my shortlist.

closed 2020-07-05 by RussTedrake