RobotLocomotion/drake #13549: Add sugar functions in MBP to remove geometries

huihuaTRI, 2020-06-11:
This should wait until [PR-13545](https://reviewable.io/reviews/robotlocomotion/drake/13545#-) gets merged. Some APIs can be defined as
- Remove the geometry with `id` associated with the `body`
`RemoveBodyGeometry(const RigidBody& body, const GeometryId& id)`
- Remove all the `role` geometries associated with the given `body`
`RemoveBodyGeometries(const RigidBody& body, const geometry::Role& role )`

SeanCurtis-TRI, 2020-06-11:
This isn't relevant. Once #13545 merges, MBP won't really know anything about geometries. And, after finalization, won't be able to serve as a sugar interface to SG. So, this sugar is neither necessary nor possible.

closed 2020-06-11 by SeanCurtis-TRI

huihuaTRI, 2020-06-11:
How about the `RigesterVisualGeometry()` sugars in MBP already,  are you going to remove those?

SeanCurtis-TRI, 2020-06-12:
None of the `Register*Geometry` sugar is going away. The difference is that *right* in master, they're required. After the PR they will purely be sugar. Because of that change of state, the use of them is ubiquitous and the convenience is non-trivial. I don't see that sugar going away any time soon.

huihuaTRI, 2020-06-12:
The closure: talked with Sean f2f offline (actually "online" via VTC). I missed the fact that MBP detaches the scene_graph pointer during the `Finalize` process, therefore, remove geometries after finalizing won't be possible. Thanks!