RobotLocomotion/drake #13552: Prevent false remote cache hits between builds on different operating systems

jamiesnape, 2020-06-12:
Relates #13102.

Per https://github.com/bazelbuild/bazel/issues/4558 and https://github.com/bazelbuild/bazel/issues/3320 seems creating separate caches is the only truly safe solution.

jamiesnape, 2020-06-12:
A minimal set of tools for a cache key would be:

* `CC` (`clang` or `gcc`)
* `gfortran`
* `java`
* `python3`

Something like

```bash
dpkg-query --showformat='${Version}' --show $(dpkg-query --search $(readlink -f $(command -v ${tool_name})) | cut -d':' -f1)
```

will get the version in a non-tool-specific way on Ubuntu (modulo some double quotes).

jwnimmer-tri, 2020-06-17:
Is this a blocker for #13434?

jamiesnape, 2020-06-17:
Pretty much. It is not a difficult task, though.

jamiesnape, 2020-07-01:
Closed by https://github.com/RobotLocomotion/drake-ci/commit/7a41caeb69722af80d9e069549db1e1184b3f506.

closed 2020-07-01 by jamiesnape