RobotLocomotion/drake #13560: Add caching for MultibodyTree::CalcAllBodyBiasSpatialAccelerationsInWorld()

mitiguy, 2020-06-15:
While improving efficiency on PR #13453, we thought caching may be useful if an end-user is doing a significant number of bias accelerations calculations -- no reason to call MultibodyTree::CalcAllBodyBiasSpatialAccelerationsInWorld() so many times if many different user-defined bias acceleration calculations are needed.

amcastro-tri, 2020-06-16:
could you specify what quantity exactly you'd cache?

mitiguy, 2020-06-16:
Current plan is to cache each body's spatial **bias** acceleration measured in world and expressed in world, with respect to speeds v.

Per conversation with @sherm1 and @rpoyner-tri, caching issue #13560 has been changed to high priority - after PR #13453 merges.  Per Sherm, there was an associated heap allocation performance issue raised by Evan Drumwright.

mitiguy, 2020-06-17:
Per suggestion by @sherm1 in PR #13453, reverted this to high priority.

jwnimmer-tri, 2024-11-01:
This issue does not explain why it's relevant, nor any benchmark that demonstrates why any speed-up is required.  Closing a "not planned".

closed 2024-11-01 by jwnimmer-tri