RobotLocomotion/drake #13565: Drake uses int for Eigen::Index at multiple places.

huihuaTRI, 2020-06-16:
In `math/rigid_transform.h`, two places ([place 1](https://github.com/RobotLocomotion/drake/blob/a05c9f898cbcd41c72319b3a89b1c741dc554726/math/rigid_transform.h#L226) and [place 2](https://github.com/RobotLocomotion/drake/blob/a05c9f898cbcd41c72319b3a89b1c741dc554726/math/rigid_transform.h#L600)) use `int` for type `Eigen::Index`, which is a type of [`std::ptrdiff_t`](https://en.cppreference.com/w/c/types/ptrdiff_t)

There will be a warning that complains about converting `long` to `int` may lose information in a 64bit platform. Drake currently ignores that warning.

It is likely that there may be other places in Drake doing the same thing.

jwnimmer-tri, 2020-06-17:
Can you give an example of the warning, and how to reproduce it?  We've been doing this forever, and I don't recall any current warnings in CI.

huihuaTRI, 2020-06-17:
I have sent you a private slack message about the example since it involves some proprietary information.

jwnimmer-tri, 2020-06-17:
The TL;DR of the thread is that this is GCC 7, with `-Werror=conversion` enabled.  That is a noisy warning that does more harm than good.  In our current support matrix, is it not enabled in Drake CI, and so downstreams enable it at their own risk.  Setting it to be a hard error in Drake code when Drake is used only an external is very aggressive and unlikely to work.

jwnimmer-tri, 2020-06-17:
I believe the conclusion from private discussions was to not enable `-Wconversion` when building Drake.  So we can close this issue?

huihuaTRI, 2020-06-17:
Is there a verdict about whether we should submit a PR to change the `int` to `Eigen::Index`? I think this issue is more about that.

sherm1, 2020-06-17:
There are hundreds of changes in Drake that would be required to get rid of all the warnings, with no real-world benefit. If changing just the two cases you encountered would significantly improve your life it would be fine to do so (although I'd just add a `static_cast<int>` rather than propagate the `Eigen::Index` any further). But likely you'll encounter the same problem when some other header gets included -- Drake isn't -Wconversion safe and likely won't ever be.

huihuaTRI, 2020-06-17:
I agree. The solution of my problem is not just changing these two places. 

Since we have a consensus that it is not necessary to make these changes for Drake's sake as Drake is not -Wconversion safe anyway, I will close this issue.

closed 2020-06-17 by huihuaTRI