RobotLocomotion/drake #13567: Clarify stribeck friction models used in multibodyplant

RussTedrake, 2020-06-17:
Our [documentation](https://drake.mit.edu/doxygen_cxx/group__stribeck__approximation.html) suggests we use a piecewise polynomial approximation, with zero derivative at the origin (that seems odd, and @sherm1 confirms it was a hold-over from simbody).  `MultibodyPlant<T>::StribeckModel::ComputeFrictionCoefficient` does have this model.

The TAMSI paper suggests that TAMSI is just using the the piecewise-linear approximation (scanning the code makes me think that's true, but it's a little hard for me to disentangle).  @antequ ?

It's also not clear to me which model is being used when.  Could we clarify?

antequ, 2020-06-22:
Also tagging @amcastro-tri 

We use two different models depending on whether the MBP is initialized in discrete vs continuous mode. Quadratic in discrete, fifth-order Stribeck in continuous.

- In discrete mode, we use the TAMSI solver, which uses the quadratic approximation with mu_s = mu_k. (so f(v) = v(2-v) for v < 1, 1 otherwise. See [TamsiSolver::RegularizedFriction()](https://github.com/RobotLocomotion/drake/blob/cf449f01fcecb328b7e7dd21f295229b9835d1ca/multibody/plant/tamsi_solver.cc#L764).) It's different from the paper, which uses a linearized version; I thought this was discussed in this documentation [here](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_tamsi_solver.html), but it looks like it's no longer there.

- In continuous mode, we use a fifth-order polynomial Stribeck approximation. See [MultibodyPlant::StribeckModel::ComputeFrictionCoefficient](https://github.com/RobotLocomotion/drake/blob/cf449f01fcecb328b7e7dd21f295229b9835d1ca/multibody/plant/multibody_plant.cc#L3019). This also seems to have been removed from the documentation.

amcastro-tri, 2020-06-24:
late to the party. I agree with your intuition @RussTedrake. The reason I designed TAMSI with a non-zero derivative functional form of the regularized friction is because I also thought it was odd. The reality is that in the stiction region we want regularized friction to behave as a very thick viscous fluid. That only happens if the derivative is non-zero at the origin. Mathematically, TAMSI will do better with a non-zero region because it informs better the search direction during our Newton-Raphson iterations. 

Now, whether quadratic or linear work better is hard to say. Thus far I have nothing conclusive tough back in the day when I wrote TAMSI in the span of two weeks for some reason I decided to use quadratic (if I only remembered!).

My most recent experiments with the continuous integrators show as much as a 30% speed up when using the coefficient with zero derivative at the origin. I believe however this is a bug. The explicit integrators I was using mistakenly believe they converge, however most often miss the stiction region. If you believe you are in sliding, you have a smoother set of ODEs, therefore your integrator has a better time. Nothing conclusive, but just an observation as I played with it.