RobotLocomotion/drake #13571: style: Should add basic linter for Python import statements

EricCousineau-TRI, 2020-06-18:
Should formalize import statements

Some items:
* From PEP008, should delineate builtin, third, and first-part sources
* Should handle ensuring `pydrake` is imported before `torch` (for Anzu)

My preferred styles:
```py
import module
from module import SingleSymbol
from package.sub_package import module
# import package.sub_package.module as module  # Nah.
from yyy_module import YyySymbol1, YyySymbol2
from zzz_module import (
   ZzzSymbolA,
   ZzzSymbolB,
   ZzzSymbolC,
   ZzzSymbolD,
)
```

Relates #13461 (PR)

jwnimmer-tri, 2020-06-18:
Using a community tool would be best, if there is a suitable one.  First search shows https://github.com/timothycrosley/isort as a contender.

jwnimmer-tri, 2022-02-14:
If I'm reading the intention here correctly, then FYI in Anzu we have an implementation of this already.  We could post that code somewhere for someone to adapt it to Drake, if there was interest.

svenevs, 2024-01-24:
xref as likely related via `isort` https://github.com/RobotLocomotion/drake/issues/19827

jwnimmer-tri, 2024-01-25:
Here's some `pyproject.toml` settings for isort that might be relevant:

```
[tool.isort]
# For more information:
# - https://timothycrosley.github.io/isort/
# - (Permalink) https://github.com/timothycrosley/isort/blob/5.1.3/README.md

# Based on `profile = black`, but with customizations.
force_grid_wrap = false
force_sort_within_sections = true
honor_noqa = true
include_trailing_comma = true
multi_line_output = 3  # Vertical Hanging Indent
use_parentheses = true
# We turn this off because we want symbols like LCM to be simple sorted.
order_by_type = false
case_sensitive = true

# [ Sections ]
# See: https://github.com/timothycrosley/isort#custom-sections-and-ordering
# This determines the ordering of imports.
sections = [
    # {builtin}
    "FUTURE",
    # {builtin} + extra_standard_library
    "STDLIB",
    # known_third_party
    "THIRDPARTY",
    # known_first_party
    "FIRSTPARTY",
    # {builtin}
    "LOCALFOLDER",
]
default_section = "THIRDPARTY"

known_first_party = [
    "drake",
    "pydrake",
]
```

jwnimmer-tri, 2024-01-25:
Also a reminder from f2f: we don't need a big-bang PR that adds linters and fixes everything.  Instead, we should PR the tool and one 1 sample subdir that enables it.  Then, incrementally we'll work on getting more and more parts of the tree to pass.

jwnimmer-tri, 2024-01-29:
The import order linter & auto-fixer imagined here will prescribe exactly how the imports should look.

In terms of choosing a configuration file for that tool, the first-order goal should be to match (as closely as reasonable) the existing import ordering that Drake's code already uses.  Of course some existing code will be wrong/inconsistent and the linter will end up fixing those files, but the first stab at config file settings should be ones that have as small of diffs as practical.

jwnimmer-tri, 2025-03-28:
It's possible that instead of just doing an import sorter, we should go straight to a complete autoformatter; see #19827.