RobotLocomotion/drake #13580: RgbdSensor.X_WB_output_port() is `PoseVector`, not `Value<RigidTransform>` :(

EricCousineau-TRI, 2020-06-22:
Docs:
https://github.com/RobotLocomotion/drake/blob/5f4d84b44b69c63a90e88b26d0d07a152b518343/systems/sensors/rgbd_sensor.h#L187-L190

Implementation:
https://github.com/RobotLocomotion/drake/blob/5f4d84b44b69c63a90e88b26d0d07a152b518343/systems/sensors/rgbd_sensor.cc#L73-L74

:crying_cat_face: 

@SeanCurtis-TRI I'm thinking I was the one to review this PR, and didn't catch this?
FYI @RussTedrake Not sure if you're using this port directly.

EricCousineau-TRI, 2020-06-22:
Retrospective: https://github.com/RobotLocomotion/drake/pull/11704#issuecomment-647227616
Dern...

EricCousineau-TRI, 2020-06-22:
I *though* the workaround would be to change above Python code to:
```py
X_WB = sensor.X_WB_output_port().EvalAbstract(context).get_transform()
X_WC = X_WB @ sensor.X_BC()
```
... but it looks like the value being output is `Value[BasicVector_[float]]`, *not* `Value[PoseVector]`?

Anywho, the workaround will just be:
```py
blarg = X_port.Eval(context)
X_BC = PoseVector(blarg).get_transform()
```

EricCousineau-TRI, 2020-06-22:
*WRONG*, `PoseVector(ndarray[7])` does not exist. Instead, I have to do:
```py
blarg = sensor.X_WB_output_port().Eval(context)
blarg_2 = PoseVector()
blarg_2.SetFromVector(blarg)
X_WB = blarg_2.get_transform()
```

:(

SeanCurtis-TRI, 2021-07-30:
In deprecating `PoseBundle` and its associated classes (including `PoseVector`), `RgbdSensor` now has the `body_pose_in_world` output port. It is `RigidTransformd` valued.

PR #15534 (2021-08-06): Deprecating PoseBundle
This deprecates `PoseBundle` as well as closely affiliated classes: `PoseVector`, `PoseAggregator`, `PoseBundoeToDrawMessage`.

There are two systems that have `PoseBundle`-valued output ports and two diagrams that export those ports.

The systems had explicit getters for the ports; those getters have been deprecated.

The diagrams were mixed. One had a getter, one relied on `get_output_port(name)`. To generally aid the case where a leaf output port is exported, the calc methods of the two systems have been given a "log once" warning.

Bindings and unit tests have likewi [... cut]
closed 2021-08-06 by jwnimmer-tri