RobotLocomotion/drake #13585: //common:py/cpplint_wrapper_test times out on Focal

BetsyMcPhail, 2020-06-22:
For example: https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=1306590

rpoyner-tri, 2020-06-23:
Adding team label as part of platform-review rounds.

jwnimmer-tri, 2020-08-05:
Running the cpplint_wrapper locally, it's somewhat broken anyway (complains about valid code, and assumes we still have a top-level "drake" folder).  Given the bazel scheduler for individual linting, I don't think this tools is useful anymore.

@BetsyMcPhail could you open a PR to remove cpplint_wrapper (and its test) from drake?  It is neither installed, nor mentioned in the documentation, so I don't think we need any deprecation period.

DamrongGuoy, 2020-08-06:
I confirm that `//common:py/cpplint_wrapper` has nothing to do with CLion instruction for "Lint selected file for google style guide": 
[https://drake.mit.edu/clion.html#lint-selected-file-for-google-style-guide](https://drake.mit.edu/clion.html#lint-selected-file-for-google-style-guide)

PR #13853 (2020-08-10): Remove cpplint_wrapper and associated test
Resolves #13585

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13853)
commit 99ef3fe6 (2020-08-10): Remove cpplint_wrapper and associated test (#13853)

* Remove cpplint_wrapper and associated test

Resolves #13585
closed 2020-08-10 by SeanCurtis-TRI