RobotLocomotion/drake #13586: Compute Hessian of MBP dynamics function

hongkai-dai, 2020-06-22:
If I have a function f(q, v, v̇, λ) = M * v̇ + C(q, v) - Bu - J'λ, I wish to compute the Hessian of this function `f` w.r.t `q, v, v̇, λ`. Some of the Hessian terms are easy to compute, but to compute ∂² f / ∂ q², it requires computing the Hessian of ∂² M / ∂ q² and ∂² C / ∂ q², which I don't know how to do yet.

I guess the easiest solution might be to instantiate MBP with nested AutoDiffScalar.

@sherm1

sherm1, 2020-06-22:
Since f is a vector function, ∂f / ∂q is a matrix. That makes ∂²f / ∂q² a hypermatrix. Is that what you're looking for?

hongkai-dai, 2020-06-22:
I should have been more clear, I mean ∂²f(i) / ∂q², namely the second order derivative of the i'th entry of `f` w.r.t `q`.

sherm1, 2020-06-22:
We don't have a way to calculate just a single generalized force f(i) so the nested AutoDiff calculation would give { ∂²f(i)/∂q² | i ∈ 0..n-1 }. (That's a way to represent the 3D hypermatrix.) I'm not seeing an easier way to get that though.

You don't have u in the functional dependence of f. Should it be there?

hongkai-dai, 2020-06-22:
> You don't have u in the functional dependence of f. Should it be there? 

Yes `u` should be there also.

> We don't have a way to calculate just a single generalized force f(i) so the nested AutoDiff calculation would give { ∂²f(i)/∂q² | i ∈ 0..n-1 }. (That's a way to represent the 3D hypermatrix.) I'm not seeing an easier way to get that though.

Makes sense. I think if we use nested autodiff, we can still return `f` as an Eigen::Vector<NestedAutoDiff>, where the i'th entry of this `f` contains the value of `f`, the gradient and the Hessian.

sherm1, 2020-06-23:
Yes, each entry would contain its value, gradient, and Hessian.
Added u in the issue description.

BTW @edrumwri wrote [a document](https://drive.google.com/file/d/1pTG_eY0-GqwGYL1IAfKNzf9bGmXpuKfM) on AutoDiffing that has a Hessian section.

Assigning to @rpoyner-tri for now as vaguely related to #10991 -- this is an application that would _really_ stretch AutoDiff performance!

rpoyner-tri, 2022-03-18:
Given where our thinking about autodiff is going, I'm handing this back to @hongkai-dai for further consideration.

cohnt, 2025-05-06:
This is something @shrutigarg914  I would find very useful -- there are various cases where we might want to compute the [kinematic Hessian](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=240619), which this would encompass.