RobotLocomotion/drake #13591: Provide copy_on_write_ptr for use in Drake

sherm1, 2020-06-23:
We have had many discussions about using copy-on-write as an optimization for dealing with large shared objects that are rarely written on (e.g. #9501). Currently we can't experiment with that approach due to lack of a suitable thread-safe smart pointer.

We have copyable_unique_ptr and should add a compatible copy_on_write_ptr. Design thoughts:
- should be thread safe by default (an atomic_int ref count should be adequate)
- should offer a non-threaded option, at least for measurement of threading cost
- read access should have no overhead (note: reading a shared_ptr does have overhead)
- borrow std::shared_ptr API where possible, e.g. `unique()`
- can't derive from copyable_unique_ptr, but could embed one

Notes
- [Herb Sutter on COW](http://www.gotw.ca/publications/optimizations.htm) 20 years ago
- A thoughtful but excessive (and LGPL) implementation [on GitHub](https://github.com/HadrienG2/copy-on-write-ptr)

jwnimmer-tri, 2020-07-07:
On a workflow note -- I would want to see a benchmark showing improvement (off-master is fine) before merging this container into the `drake/common` folder on master.  (Rationale: we shouldn't invest time in platform review -- and maybe not even feature review -- until we know that it will pay off for use cases that we care about.)  We can enable _experiments_ with the COW pointer via a quick local prototype, or using the LGPL library noted above, or at worst writing our own in a `systems/framework/dev` folder or similar on master.  The COW pointer is a means to an end, not an end in and of itself.

sherm1, 2020-07-07:
That workflow makes sense to me. I need help finding an application in which unnecessary copying of large objects (e.g. GeometryState) is a performance problem. As possible example, is anyone saving trajectories as an `std::vector<Context>`? Are there other applications where read-only sharing would make a noticeable difference?

jwnimmer-tri, 2024-03-20:
This is a fine idea, but "not planned" until we have some specific need for it.

closed 2024-03-20 by jwnimmer-tri