RobotLocomotion/drake #13597: [geometry] ConnectDrakeVisualizer ignores geometry state in context

SeanCurtis-TRI, 2020-06-25:
Calling `ConnectDrakeVisualizer` adds some systems to the diagram which are responsible for dispatching LCM message to initialize and update the illustrated SceneGraph geometries in `drake_visualizer`. Unfortunately, when the initialization event is processed to generate the "load" message, the system queries [the model inspector for the `SceneGraph` whose pointer has been stored](https://github.com/RobotLocomotion/drake/blob/master/geometry/geometry_visualization.cc#L283-L284) rather than using the world as described in the context. It should be implemented to depend on the context.

relates #13126

EricCousineau-TRI, 2020-11-03:
See also: #10482

closed 2020-11-06 by SeanCurtis-TRI

jwnimmer-tri, 2023-07-31:
Several places in the code still mention this issue.  Possibly those should be cleaned up.

jwnimmer-tri, 2024-02-17:
> Several places in the code still mention this issue. Possibly those should be cleaned up.

=> https://github.com/RobotLocomotion/drake/issues/20962

PR #14282 (2020-11-06): [geometry] Deprecate geometry_visualization (and old ConnectDrakeVisualizer)
This PR accomplishes two things:

  1. It extends the interface on `DrakeVisualizer` to add some sugar comparable to the current `ConnectDrakeVisualizer`.
  2. It deprecates the public API in `geometry_visualization.h` and replaces all calls with the new API.

The PR has been curated in two commits: one for each of the items above.

Resolves #13597

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14282)