RobotLocomotion/drake #13603: document the fact that gurobi and mosek are not available in the binary builds

RussTedrake, 2020-06-27:
In responding to this question:
https://stackoverflow.com/questions/62587294/nightly-build-doesnt-acquire-mosek-license/62613736#62613736
I went to look for the place in the documentation, or an open issue, that I could cite.  I didn't find one.

https://drake.mit.edu/doxygen_cxx/group__solvers.html says SNOPT is provided in the binary release, but I didn't see anywhere we say that Gurobi and Mosek are not.

Is it written somewhere?  If not, can we add it? 
cc @hongkai-dai

hongkai-dai, 2020-06-27:
No I don't think it is written anywhere. We should probably add it in https://drake.mit.edu/from_binary.html

jwnimmer-tri, 2020-06-29:
I've added this to the #13539 Epic.

jamiesnape, 2020-06-29:
As far as the Stack Overflow question goes, I guess `pydrake.solvers.mosek.MosekSolver.AcquireLicense().is_valid()` returning `False` is technically correct, but maybe it is not the most user friendly outcome?

ToffeeAlbina-TRI, 2020-07-17:
@RussTedrake this is captured under umbrella #13539 which is on deck on the [build system, continuous integration, distribution](https://github.com/RobotLocomotion/drake/projects/2) board, so I'm removing it from the [improve pydrake documentation *user* experience](https://github.com/RobotLocomotion/drake/projects/4) board.

jwnimmer-tri, 2021-11-10:
The place to mention this is probably the four pages linked from https://drake.mit.edu/installation.html (at the "Stable" links), as well as the https://drake.mit.edu/doxygen_cxx/group__solvers.html overview.

closed 2022-01-11 by jwnimmer-tri