RobotLocomotion/drake #13606: Additional Focal Dependencies for Binary Install

iperper, 2020-06-28:
Recently upgraded from 18.04 to 20.04 via clean install. 

Today installed Drake via binary [https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz](https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz).

Added drake to python path. 

`import pydrake` => `ImportError: libClp.so.1`.

This was because I forgot to run `sudo .\install_prereqs`.  After running script, first error resolved.

Try again: `import pydrake` => `ImportError: libnlopt.so.0`. Resolved with `sudo apt install libnlopt0`

Try again: `import pydrake` => `ImportError: libOpenGL.so.0`. Resolved with `sudo apt install libopengl0`

Didn't have this issue with 18.04, although likely because these libraries were available on my system from all the other projects over time.  Not sure if `sudo apt install libopengl-dev` is preferred over non-dev version.  I don't think I missed any install steps (running setup script and adding to Python path).

jwnimmer-tri, 2020-06-29:
Hi @iperper, thanks for your report!

It seems like you're using Ubuntu 20.04.  However ...

> Today installed Drake via binary https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz.

... this link doesn't seem correct, for that case.  Could you share which link you were using?

I'll investigate the dependencies, in any case.

iperper, 2020-06-29:
Oops.  Sorry copied the wrong link.  It was [https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-focal.tar.gz](https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-focal.tar.gz)

jamiesnape, 2020-07-01:
Looks like we are querying the wrong pkg-config file for nlopt (C++) on focal, so we are linking to `libnlopt0` (C) when we probably do not need to. On Bionic, there was no split between C and C++. For OpenGL, looks like we are probably installing `libGL.so` and not `libOpenGL.so`. Linking to OpenGL.so was intentional, so that is correct.

jamiesnape, 2020-07-01:
```
$ ldd libdrake.so | grep "not found"
	libnlopt.so.0 => not found
	libOpenGL.so.0 => not found
	libOpenGL.so.0 => not found
$ ldd --unused libdrake.so | grep nlopt     
	libnlopt.so.0
```

(Aside: I should probably fix `ldd --unused libdrake.so` one day. We are over-linking a lot.)

jamiesnape, 2020-07-02:
I may as well take this.

PR #13649 (2020-07-16): Fix and/or cleanup OpenGL dependencies
Closes #13606.

---

We need either `libOpenGL.so`, `libOpenGL.so.0`, `libGLX.so`, `libGLX.0`, `GL/gl.h`, and
`GL/glx.h` (modern, vendor neutral) or `libGL.so`, `libGL.so.1`, `GL/gl.h`, and `GL/glx.h`
(legacy). Vendor neutral is preferable, for NVIDIA users and others, so we choose `libOpenGL.so` and `libGLX.so`
rather than `libGL.so`.

Also rebuild VTK and drake-visualizer to ensure they are correctly linked to
`libOpenGL.so` and `libGLX.so` rather than `libGL.so`.

**Ubuntu 18.04 (Bionic)**

*package contents:*

| package           | libraries           [... cut]
closed 2020-07-16 by jamiesnape