RobotLocomotion/drake #13609: VPN issues with Bazel and CMake

mfry90, 2020-06-29:
When following installation instructions for python bindings (https://drake.mit.edu/python_bindings.html), cmake is used as a wrapper around bazel:
```
cmake ../drake
make -j
```
I have some problems with building from Bazel due to my VPN (apparently this is a known issue to Bazel, https://github.com/bazelbuild/bazel/issues/3602), but when calling Bazel directly, the following works fine:
```
bazel --batch build //...
```

When relying on the CMake wrappers, is there a way I could set a CMake flag to pass the --batch flag onto bazel?

jwnimmer-tri, 2020-06-29:
Does it work if you create a `$HOME/.bazelrc` file with this line as its content?
```
startup --batch
```

My local testing suggests that this might be sufficient.

Since this is presumably a machine-wide problem, fixing it in a homedir dotfile (or even in `/etc/bazel.rc`, if you prefer) seems more suitable that a CMake option.

See https://docs.bazel.build/versions/master/guide.html#bazelrc for details on rcfiles.

mfry90, 2020-06-29:
Yes, that does it, thanks!

jwnimmer-tri, 2020-06-29:
Awesome!

closed 2020-06-29 by jwnimmer-tri