RobotLocomotion/drake #13614: relocate content from lingering .tex files in doc/ directory

RussTedrake, 2020-06-30:
We have a few .tex files in the drake/doc directory that I suspect aren't in the best place.  @hongkai-dai -- I believe both are yours.  Consider moving that information and zapping the files?

hongkai-dai, 2020-07-04:
There are several `tex` files in drake/doc folder, including

1. [aerodynamics.texp](https://github.com/RobotLocomotion/drake/blob/master/doc/aerodynamics.texp). I believe this is outdated as it still uses `RigidBodyWing`, we should zap it.
2. [doc_cartesian2_cylindrical.tex](https://github.com/RobotLocomotion/drake/blob/master/doc/doc_cartesian2cylindrical.tex) (together with doc_cartesian2cylindrical.pdf, cartesian2cylindrical.svg). This is outdated. I created them back in DRC days. I will zap them.
3. [ik.tex](https://github.com/RobotLocomotion/drake/blob/master/doc/ik.tex) This is for the IK engine we used in MATLAB. I will zap it.
4. [pivot_column.tex](https://github.com/RobotLocomotion/drake/blob/master/doc/pivot_column.tex) This is for the Lemke solver still used in Drake now. Should I move it to `solvers/doc` folder?
5. [sw2urdf.tex](https://github.com/RobotLocomotion/drake/blob/master/doc/sw2urdf.tex) I believe this was created during the DRC days. I am not sure if it is still useful as I don't have a Windows machine with solidworks on it.

RussTedrake, 2020-07-04:
sounds good on all of the zapping.  I don't mind leaving pivot_column.tex in the doc directory if we want to keep it in drake.  It just seems like nobody will find it there?  Would moving it into doxygen be better?

hongkai-dai, 2020-07-04:
Evan and I used a lot of math symbols in pivot_column.tex, that makes it hard to write with unicode, hence I am afraid it is hard to move it into doxygen? I think pivot_column.tex is more like a technical report on the Lemke solver we implemented, and we kept it in drake for future maintenance on unrevised_lemke_solver.h/cc, as mentioned in https://github.com/RobotLocomotion/drake/blob/a0f8ae243d2fbdc116bb676276714ea90d1d3eb9/solvers/unrevised_lemke_solver.h#L30-L35, not as a public documentation for Drake users. Do you think it should live in doxygen as other public documentation?

RussTedrake, 2020-07-04:
In that case, I'm ok with it staying where it is.

closed 2020-07-06 by hongkai-dai