RobotLocomotion/drake #13617: [render] RenderEngineGl requires missing features

SeanCurtis-TRI, 2020-06-30:
We've ported `RenderEngineGl` from Anzu back to Drake (#13584). However, the ported code is incomplete:

- [ ] The ability to display the rendering is missing.
- [ ] Unable to render label images.
- [ ] Unable to render color images.
- [ ] Needs to support full intrinsic specification (see, e.g., #11880, #13557).

This epic PR tracks the work for the `RenderEngineGl` to be "complete" (in the sense that it has the same functionality as the other `RenderEngine` implementations).

jamiesnape, 2020-07-07:
When I read the title, I thought that it mean RenderEngineGl required missing features of a certain OpenGL version. Do you mean "RenderEngineGl is missing (required) features"?

SeanCurtis-TRI, 2020-07-07:
@jamiesnape Yep. You're dead on. I've tweaked the title so it isn't so confusing.

SeanCurtis-TRI, 2020-11-16:
Opening this back up to make sure we add `RenderEngineGl` to the render benchmarks.

PR #14289 (2020-11-10): [rendering] Give RenderEngineGl access to full camera intrinsics
 - Refactor the mapping from `RenderCamera` --> OpenGl Projection matrix previously defined in `RenderEngineVtk` into `RenderCameraCore`.
 - Add convenience constructors for `*RenderCamera` instances from `CameraProperties` instances.
 - Add `RenderCamera`-based rendering to `RenderEngineGl`
   - Implement `DoRender*Image(*RenderCamera)` API from `RenderEngine`
   - Express `Render*Image(*CameraProperties)` in terms of the `DoRender*Image` API. This is about 95% cut-and-paste and 5% changing `CameraProperties` declarations to `Re [... cut]
PR #14540 (2021-01-21): [render] Add RenderEngineGL to the render engine benchmark
Briefly, this adds `RenderEngineGl` to the render engine benchmark.

As part of that activity, it does several other things:

  - Move the documentation out of the .cc file into a new .h file so that it appears in the rendered documentation.
  - Refactor the benchmark code to allow more strategic benchmarking.
  - Correct oversight in `RenderEngineGl` implementation (posing light now works).

closes #13617

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14540)
closed 2021-01-21 by SeanCurtis-TRI