RobotLocomotion/drake #13624: jaco_description lacks package.xml

ggould-tri, 2020-07-01:
Most of the manipulation models subdirectories have package.xml files to allow ROS-style access via the package:// URI scheme.

> ggould@tars:~/git/drake$ ls manipulation/models/*/package.xml
> manipulation/models/allegro_hand_description/package.xml
> manipulation/models/franka_description/package.xml
> manipulation/models/iiwa_description/package.xml
> manipulation/models/wsg_50_description/package.xml

Note that the `jaco_description` (and `ycb`) subdirs are not on that list.

This required a fairly ugly workaround in a downstream project.

FYI @sammy-tri @EricCousineau-TRI

jwnimmer-tri, 2020-07-01:
For the record, we have discussed removing _all_ of those `package.xml` files in lieu of a single `package.xml` at the root of Drake.  But until we make that change elsewhere, we should be consistent for the jaco.

jwnimmer-tri, 2020-10-01:
Fixed by #14134 perhaps?

EricCousineau-TRI, 2020-10-01:
Oh yeah, oops!

closed 2020-10-01 by EricCousineau-TRI