RobotLocomotion/drake #13627: [render] Image rendering speed is too slow

sherm1, 2020-07-01:
From email exchange with @dhelmick-tri:
> ... I did want to highlight what I think is currently our biggest pain point with the sim though: the image rendering speed.  Right now it takes sim about 500ms to render a single ~5MP RGB image.  So I just wanted to say that it would have a very large impact on the effectiveness of our team if we could reduce that time significantly.  Just to put a bound on the request, in an ideal world, I don't see us needing, anytime in the near future, anything more than being able to render 2x ~5MP RGB images at > 10 Hz.  This would be 50ms/image, which is only an order of magnitude faster than the current rendering speed. :) ...
Our perception pipeline does use both CPU and GPU processing so we do copy our images from system memory to GPU memory (multiple times per cycle actually).  We've profiled that though and it only takes 5-6 ms to copy 2x 5MP RGB images from system to GPU memory, so I don't think that's going to be your tall tent pole.

SeanCurtis-TRI, 2021-01-19:
As of #14540, the render benchmark has been updated to include `RenderEngineGl`. As relates to this issue:

`RenderEngineVtk`:
  - For a single geometry: color 2560x1920 time is 11.9 ms.
  - For 1200 geometries: color 2560x1920 time is 750 ms.
  - For 1200 geometries, color even 320x240 time is 654 ms.
`RenderEngineGl`:
  - For a single geometry: color 2560x1920 time is 11.4 ms.
  - For 1200 geometries: color 2560x1920 time is 14 ms.
  - For 1200 geometries, color even 320x240 time is 3.75 ms.

Conclusion:
  - `RenderEngineVtk` has a *huge* per-geometry cost. A cost that falls off a cliff when too many geometries get added. That per-geometry cost absolutely swamps the cost of processing the image.
  - I'm forced to infer that the scenes under question have many geometries.
  - Generally, it seems that everything will be better if the user simply swaps to `RenderEngineGl`.

sherm1, 2021-01-19:
Awesome! Thanks, @SeanCurtis-TRI.

sherm1, 2021-01-19:
/cc @huihuaTRI

huihuaTRI, 2021-01-19:
Thank you for the update!  I will let the FL team know about it. They will need to update their Drake version though.

SeanCurtis-TRI, 2021-12-04:
Closed as no obvious fix.

closed 2021-12-04 by SeanCurtis-TRI