RobotLocomotion/drake #13639: Crash in Drake Visualizer in a docker

cjds, 2020-07-04:
> Before you post:

I'm getting a segmentation fault while trying to run the Drake Visualizer from a Dockerfile. This is a relevant dockerfile that replicates the issue

```
FROM robotlocomotion/drake
RUN apt update \
    && apt install -y curl gnupg software-properties-common \
    && curl https://bazel.build/bazel-release.pub.gpg | apt-key add - \
    && echo "deb [arch=amd64] https://storage.googleapis.com/bazel-apt stable jdk1.8" | tee /etc/apt/sources.list.d/bazel.list \
    && apt update \
    && apt install -y bazel
RUN add-apt-repository ppa:neovim-ppa/stable \
    && apt-get update \
    && apt-get install -y neovim
RUN apt-get install -y coinor-libipopt-dev \
                       libsuitesparse-dev \
                       libtinyxml-dev \
                       libblas-dev \
                       liblapack-dev \
                       libnlopt-dev \
                       gfortran
```

This is the run command

```
xhost +
docker run -itd --privileged \
           --env DISPLAY \
           --net=host \
           -v $PWD:/root/doublependulum \
           -v ~/.vim:/root/.vim \
           -v ${HOME}/.Xauthority:/home/user/.Xauthority \
           --device /dev/dri \
           --name drake drake bash
```

This is a Core dump
https://filebin.net/95fkjojxarojcfbe

jamiesnape, 2020-07-07:
Thank you for the report.

Unfortunately, this is not a configuration that we test, and [we have had discussions](https://github.com/RobotLocomotion/drake/issues/12483#issuecomment-635350843) about whether we can even support drake-visualizer running in a Docker container on a wide enough variety of platforms. I will certainly take a look at the stack trace, but the resolution may be an update to our documentation clarifying that this configuration is not supported.

What is your host operating system and what graphics card and driver are you using?

cjds, 2020-07-07:
Fair enough. I can appreciate supporting the myriad of graphics drivers on the host is complex enough.

I'm on Ubuntu 18.04 with a nvidia-440 driver. Graphics card is Quadro P2000 Mobile

jamiesnape, 2020-07-23:
@cjds When you have a spare moment, can you try `robotlocomotion/drake:focal` as the base image on the same host system and let me know if the result is different?

jamiesnape, 2020-08-11:
Closed by #13764.

closed 2020-08-11 by jamiesnape