RobotLocomotion/drake #13646: MBP composite COM methods claim to check composite_mass but don't

sherm1, 2020-07-06:
All the MBP CenterOfMass methods have this documentation:
```c++
  /// @throws std::runtime_error if `MultibodyPlant` has no body except
  ///   `world_body()`.
  /// @throws std::runtime_error unless `composite_mass > 0`.
  Vector3<T> CalcCenterOfMassPosition() 
```
I can see that `CalcCenterOfMassPosition()` actually does perform the composite_mass check. However, `CalcJacobianCenterOfMassTranslationalVelocity()` and `CalcBiasCenterOfMassTranslationalAcceleration()` don't appear to do so.

Please audit to make sure all the COM methods actually perform the checks they claim and add unit tests as needed.

mitiguy, 2020-07-08:
@sherm1  -- Got it.  I'll take a look.

mitiguy, 2020-11-11:
When it merges, I anticipate PR# 14316 resolves this issue.

mitiguy, 2020-11-17:
Closing this as PR# 14316 resolves this issue #13646.
https://reviewable.io/reviews/RobotLocomotion/drake/14316

closed 2020-11-17 by mitiguy