RobotLocomotion/drake #13647: pydrake doc: consider splitting up docs per class instead of per module

RussTedrake, 2020-07-07:
Transcribed from a meeting discussion:

JN: We currently have one page per package, instead of one page per class (e.g., [pydrake.math](https://drake.mit.edu/pydrake/pydrake.math.html) package).  Can you imagine if Doxygen had one page per namespace?  I suspect this is at least half the problem with the docs.
EC: Just cleaning up the classes shown can be solved (see #9599) with some heuristics.
SC: I have the sense that page-per-module is the norm for python documentation (e.g., https://docs.python.org/3/library/os.html).
JN: Yes it is true generally, but our modules are YUUUUGE.
SC: So is os. Python doesn’t seem to care.
EC: We could use a better Sphinx template. Ours is kinda suxorz. May use pytorch's template.

jamiesnape, 2020-07-07:
What is the specific problem we are trying to solve? Long web pages are not in themselves particularly troublesome if you have the right navigation (and some people prefer them). Single pages are bad if you cannot navigate between them effectively and find related documentation. So pages are hard to find? Poorly organized? Impossible to traverse? ...

SeanCurtis-TRI, 2020-07-07:
Perhaps the issue shouldn't be: "split up docs". The issue should be "resolve unnavigable module pages". 

Certainly, in python's docs, long pages are common and it is mitigated by a left-frame table-of contents. It can be expanded and collapsed and can be used to jump to arbitrary sections on the page. Building the per page table of contents (a la [python::os](https://docs.python.org/3/library/os.html)) would be another way to resolve the fundamental issue.

In my mind, this issue is related to @EricCousineau-TRI's:  "Ours is kinda suxorz".

jamiesnape, 2020-07-08:
To be honest, the Read the Docs theme was chosen purely for aesthetics. That being said, I am not sure the PyTorch theme is much better, but, equally, I do not really want us to have to write and maintain our own theme/templates.

> In my mind, this issue is related to @EricCousineau-TRI's: "Ours is kinda suxorz".

Yes. It does not roxxorz...

jwnimmer-tri, 2023-06-08:
Since this was posted, we do have a new theme now (#17012).  I wonder how we think things stand now?

I think I've learned to accept the long pages.  Having more detail in the sidebar navigation (e.g., a second panel with an index of functions in that module, like the ToC stuff in https://docs.python.org/3/library/) would probably still help, though.

We could probably also do more to compress non-default template instantiations, and the overload set documentation for functions.

svenevs, 2023-06-08:
What you are missing is the `localtoc` in sphinx terms.  Themes like the [sphinx-book-theme](https://sphinx-book-theme.readthedocs.io/en/stable/) will have the `globaltoc` on the left, and the `localtoc` on the right.  It is theme dependent, I'm sure it could be stubbed into the existing html theme too (which looks great!  it's just missing the extra navigation).

What is difficult though is that in order for the `localtoc` to be useful, it needs to have headings for things to show up I believe.  But the proposition is to add a navbar on the right that has hyperlinks to everything documented, which should be possible given that we're in control of how these things are generated.  You can add metadata to the document to enumerate all the items documented, and have the theme iterate that metadata and generate the navbar on the right?

IIRC @mwoehlke-kitware is well versed in sphinx as well and may have some additional insight.

mwoehlke-kitware, 2023-06-12:
> Note that for this kind of theming you do need to take care about html and sizes, on smaller screens they usually drop it to a hamburger menu since otherwise three panels is too small.

I feel like I've seen the local TOC *combined* with the global TOC. At the very least, they don't need to be in separate sidebars.

Maybe we just need https://sphinxcontrib-fulltoc.readthedocs.io/en/latest/?

YeZhao, 2023-06-13:
Hello all, it is great to see this conversation! Thanks all for supporting this effort. I'd like to share a note from the pydrake user perspective. I instructed an optimization course using underactuated robotics online resources this past spring semester. Given the feedback from my students in the class, we would love to see improved python API documentation. Sometimes it takes nontrivial time for students to read and find variable types, and they had to go back to the c++ documents to find out what exactly that python binding function is doing. Having a better search functionality would be helpful, e.g., more useful hyperlinks to the relevant sections rather than the current text-based search.

closed 2024-03-21 by jwnimmer-tri