RobotLocomotion/drake #13648: Support macOS Big Sur 11.0 on x86_64

jamiesnape, 2020-07-07:
Note for Ubuntu users, the Apple compiler, version 12.0 behaves most like Clang 10. If a build fails with Clang 10, it will probably fail on macOS Big Sur (and on Catalina once Xcode 12 is released).

For macOS Big Sur 11.0 on arm64, see #13514.

jamiesnape, 2020-07-07:
We would need to decide what to do about `-Wrange-loop-analysis` warnings/errors. They are trivial to fix, but not something that GCC or Clang 9 picks up.

jamiesnape, 2020-12-03:
Only real error I am seeing is (and this is literally a cut-and-paste without even considering the cause):
```
FAIL: //examples/manipulation_station:end_effector_teleop_sliders_test
INFO: From Testing //examples/manipulation_station:end_effector_teleop_sliders_test:
==================== Test output for //examples/manipulation_station:end_effector_teleop_sliders_test:
Traceback (most recent call last):
  File ".../execroot/drake/bazel-out/darwin-opt/bin/examples/manipulation_station/end_effector_teleop_sliders_test.runfiles/drake/examples/manipulation_station/end_effector_teleop_sliders.py", line 10, in <module>
    from pydrake.examples.manipulation_station import (
ImportError: ImportError: ImportError: ImportError: C++ type is not registered in pybind: NSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEE
================================================================================
```
FYI @EricCousineau-TRI.

EricCousineau-TRI, 2020-12-03:
Huh... That smells like `pybind11/stl.h` isn't included in `maniptulation_station_py.cc`... but that should show up in more than a specific flavor of macOS.

Also, at least in that module, it's included:
https://github.com/RobotLocomotion/drake/blob/aa4de29980976229c9d1897d0399d5c484e2ebdf/bindings/pydrake/examples/manipulation_station_py.cc#L3

Can you see about debugging this to trace which specific module this occurred in?
From the stack of `ImportError: ImportError: ImportError: ImportError:`, it looks like it's probably something in `pydrake/common`?

(Looking..)

Can you trying adding `pybind11/stl.h` to this file and see if it works?
https://github.com/RobotLocomotion/drake/blob/aa4de29980976229c9d1897d0399d5c484e2ebdf/bindings/pydrake/common/value_py.cc

jamiesnape, 2020-12-03:
Just for fun, I added `#include "pybind11/stl.h"` to every bindings file, with no luck. Will revisit some other time. I feel like looking at the libc++ source code is going to be in order.

(I selfishly wish we had libc++ Linux builds so someone else gets burned...)

EricCousineau-TRI, 2020-12-03:
ahhh, nuts! I sent you down the wrong rabbit hole, sorry!

The error actually comes from here:
https://github.com/RobotLocomotion/drake/blob/688a07c23fcb6b839ea8248f274f5b461a07311e/bindings/pydrake/common/cpp_param_pybind.cc#L104-L108

Probably just some RTTI mismatch? (private / public shindig, like what motivated #8706?)

jamiesnape, 2021-02-01:
Getting the useful error message `macOS 11 or later required !` when running on macOS 11 from five or so Python-related tests. Need to track down where that is coming from.

jamiesnape, 2021-02-01:
Probably in `tkinter`.

jamiesnape, 2021-02-01:
Ok, that is fixed in Python 3.9, but then other things are broken (basically the same as on Catalina), so I need to do the Python 3.9 upgrade either first or concurrently.

jamiesnape, 2021-02-02:
https://github.com/Homebrew/homebrew-core/pull/67378 is why Python 3.9 is working.

jamiesnape, 2021-02-02:
https://github.com/Homebrew/homebrew-core/pull/70249 might make my life a little easier.

jamiesnape, 2021-02-08:
Looks like the CI images got hit by the 11.0.1 instead of 11.1 upgrade bug, which I need to fix (not difficult, but annoying) otherwise we forever be stuck on 11.0.1 (which affects Xcode versions and other things).

jamiesnape, 2021-02-15:
All Jenkins jobs have been created, have run successfully multiple times, and are listed on the production pages.

closed 2021-02-15 by jamiesnape