RobotLocomotion/drake #13657: Add option to simulate armature inertia for revolute joints

yangwill, 2020-07-08:
I would like the ability to simulate the contribution of armature inertias for joints with high transmission ratios. A quick-and-dirty method is to simply add constant terms to the diagonal terms of the generalized mass matrix (Featherstone 2008 at the end of chapter 9.6).

MuJoCo implements this through an "armature" term in their model description (http://www.mujoco.org/book/XMLreference.html#joint). However, it might be useful to include an API to set this programmatically.

This is similar to #12335 though it ignores rotor dynamics and transmission ratios and instead works in generalized forces and velocities.

sherm1, 2020-07-08:
A few thoughts:
- This would be yet another per-dof joint parameter like damping and limits. Likely will require revisiting that API to be more extensible since there are more coming (like `reduction` in #12335). 
- To my ears, "armature" doesn't seem like a great name for this quantity -- is there a better name that would make it more obvious what this does? "added_inertia" or something like that?

amcastro-tri, 2020-07-09:
I've seen "reflected load inertia" out there.

sherm1, 2020-07-09:
"reflected load inertia" seems too good for this -- if I'm understanding correctly, it doesn't actually calculate the correct inertia but rather lets a user hack any number in there (could be an attempt at a reflected inertia estimate, could just be a regularization term for stability e.g.). In contrast a gear reduction factor as in #12335 would determine the actual reflected inertia.

amcastro-tri, 2020-07-09:
That's not how I understood this issue. I realize now it is an exact duplicate of  #12335. 
Meaning, I understood this issue as adding "gear ratio" and "motor inertia" to `multibody::JointActuator` and then let MBP update these reflected inertias as outlined in Featherstone's book.

sherm1, 2020-07-09:
Reopening -- I think this is a separate feature from #12335. For example, adding non-physical finger inertias just for increased stability could be accommodated with the "armature" feature, but not with a "gear reduction" feature (since there aren't any gears there).

yangwill, 2020-07-09:
I agree - this feature request is less about accurately modeling transmission ratios and more about having the option to modify inertias for stability reasons or as an simple way to account for reflected inertias without having to reason about gear reductions. Sorry for not making that clear in the initial description.

amcastro-tri, 2020-07-09:
> without having to reason about gear reductions.

IMO specifying a guess for he motor inertia and a (known) gear ratio is as complex of a task as specifying a single reflected inertia? That is, in terms of SysID it is exactly the same complexity of a problem the only change is a scaling factor (gear ratio squared).

Therefore, IMO I would never implement this feature, but the actual complete feature in #12335.

sherm1, 2020-07-10:
Per discussion in #12335, the "armature" equivalent would be to set motor_inertia to the armature value and gear_ratio=1. I'll close this issue since it is now subsumed by that one.

closed 2020-07-10 by sherm1