RobotLocomotion/drake #13662: Enable JacobianWrtVariable.kQdot for CalcBiasSpatialAcceleration() and CalcBiasTranslationalAcceleration()

mitiguy, 2020-07-09:
This is part of the outstanding epic with CalcBiasSpatialAcceleration() and CalcBiasTranslationalAcceleration().
It was raised again in PR #13638.

mamurtaza88, 2021-03-11:
Is there any update on this? Also is there an alternative to compute \dot{J}\dot{q} using any other mean apart from CalcBiasSpatialAcceleration? 

Any guidance will be appreciated

mamurtaza88, 2021-03-11:
when running the command, it builds fine but when we try to run, we get the following error

terminate called after throwing an instance of 'drake::internal::assertion_error'
  what():  Failure at multibody/tree/multibody_tree.cc:1575 in CalcBiasSpatialAcceleration(): condition 'with_respect_to == JacobianWrtVariable::kV' failed.
Aborted (core dumped)

mitiguy, 2021-03-11:
@mamurtaza88  -- It would be helpful (albeit not essential) to better understand your use case.  Would it be possible to schedule a time to talk to you about both your comments?  I am available later today.  If not - no worries.  We can bump this up in my priority list.

mamurtaza88, 2021-03-11:
Hi @mitiguy thank you for reaching back. I can explain this to you in detail. Let me know the medium and time and I will make sure of my availability.

mitiguy, 2021-03-11:
Hi @mamurtaza88   I set up a Google Hangout for us at 3:00 US Pacific time today.
meet.google.com/wcq-moty-faj
Alternatively, feel free to call me now (1:15-2:00) Pacific time or then: 650-346-9595  (voice-only, no text).

mitiguy, 2021-03-12:
@mamurtaza88   Thank you for taking time to provide the details.  I think a good action item on my side is to improve the documentation, e.g., how working with kV may suffice in many cases.  Thank you again.