RobotLocomotion/drake #13663: ycb objects lost their textures in drake visualizer?

RussTedrake, 2020-07-09:
Running ` bazel run //examples/manipulation_station:end_effector_teleop_sliders -- --setup=clutter_clearing` currently results in textureless YCB objects.

[image]

They used to have textures (and continue to have texture in meshcat).  I hope it's not just something stale locally.  @SeanCurtis-TRI -- perhaps you can reproduce?

SeanCurtis-TRI, 2020-07-09:
Running this locally, I get the following spew (once for each of the texture map associated with the loaded ycb models:

```
Generic Warning: In /vtk/IO/Import/vtkOBJImporterInternals.cxx, line 108
mtl file material_0 requests texture file that appears not to exist: 010_potted_meat_can_textured.png ; texture path: /home/seancurtis/.cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/drake/bazel-out/k8-opt/bin/examples/manipulation_station/end_effector_teleop_sliders.runfiles/drake/manipulation/models/ycb/meshes/

ERROR: In /vtk/IO/Import/vtkOBJImporterInternals.cxx, line 474
vtkOBJPolyDataProcessor (0x563d6b765ef0): Nonexistent texture image type!? imagefile: /home/seancurtis/.cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/drake/bazel-out/k8-opt/bin/examples/manipulation_station/end_effector_teleop_sliders.runfiles/drake/manipulation/models/ycb/meshes/010_potted_meat_can_textured.png 

ERROR: In /vtk/IO/Import/vtkOBJImporterInternals.cxx, line 474
vtkOBJPolyDataProcessor (0x563d6b765ef0): Nonexistent texture image type!? imagefile: /home/seancurtis/.cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/drake/bazel-out/k8-opt/bin/examples/manipulation_station/end_effector_teleop_sliders.runfiles/drake/manipulation/models/ycb/meshes/010_potted_meat_can_textured.png 
```

I can confirm that the image in question exists (with the obj and the mtl files). So, I have no idea why `drake_visualizer` if failing to load them.

SeanCurtis-TRI, 2020-07-09:
Trying to open the image in gimp and gimp is reporting denied permissions. But if I open the obj in the same directory with meshlab, it loads the obj *with* the texture.

__*FAILS*__
```
$ gimp /home/seancurtis/.cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/drake/bazel-out/k8-opt/bin/examples/manipulation_station/end_effector_teleop_sliders.runfiles/drake/manipulation/models/ycb/meshes/010_potted_meat_can_textured.png
```

__*WORKS*__
```meshlab /home/seancurtis/.cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/drake/bazel-out/k8-opt/bin/examples/manipulation_station/end_effector_teleop_sliders.runfiles/drake/manipulation/models/ycb/meshes/010_potted_meat_can_textured.obj
```

Not sure how to interpret this relative to permissions.

@jwnimmer-tri Do you have any insight?

jwnimmer-tri, 2020-07-09:
Somehow there is a blank space at the end of the filename `...texture.png_` vs `...texture.png`.

SeanCurtis-TRI, 2020-07-28:
@jwnimmer-tri Do you have any reason to believe that RobotLocomotion/models#11 didn't resolve this?
@RussTedrake probably worth confirming that things are better now.

jwnimmer-tri, 2020-07-28:
To resolve this, we'd at minimum need a PR to `tools/workspace/models/repository.bzl` to incorporate the update.

PR #13786 (2020-08-03): workspace: Upgrade models to latest commit
Closes #13663.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13786)
closed 2020-08-03 by jwnimmer-tri