RobotLocomotion/drake #13664: [visualization] Hydroelastic pressure fields render incorrectly

SeanCurtis-TRI, 2020-07-09:
# Problem
`drake_visualizer` ostensibly supports the display of the pressure field on the hydroelastic contact surface.

Currently, it displays like this:
[image]

However, this is a flame color map -- the colors should go from black -> blue -> red -> orange -> yellow -> white. But we're only displaying a gradient between black and white.

# Diagnosis
The pressure field visualization is done with vertex colors. The vertex colors get linearly interpolated along the triangles. Where we have coarse triangles covering large pressure gradients, we significantly undersample the color map.

# Solution
Rather than using vertex colors, we should:

  - Introduce the color map as a 1D texture.
  - Use texture coordinates at each vertex (in the range [0, 1]) to place it along that color range.

This will allow arbitrary color maps to be interpolated across the triangle because we're now, essentially, interpolating normalized depth (which maps appropriately at every fragment) rather only at the vertices.

amcastro-tri, 2020-07-10:
> The vertex colors get linearly interpolated along the triangles. 

Awesome! thanks for the explanation, makes sense. For my own education, and hopefully helps whoever ends up owning this, who does this interpolation? is it somewhere on Drake's side? in the visualizer? or one of the lil python scripts that the viz loads for hydro visualization?

SeanCurtis-TRI, 2020-07-13:
The interpolation happens as part of the triangle rasterization process -- all on the GPU. So, the person who resolves this issue is one who is familiar with OpenGL-style rasterization workflows and understands the implications of associating per-vertex data and how it manifests across corresponding triangles.

SeanCurtis-TRI, 2021-02-11:
Resolved by #14622

closed 2021-02-11 by SeanCurtis-TRI