RobotLocomotion/drake #13665: [visualization] Poor hydroelastic contact visualization performance

SeanCurtis-TRI, 2020-07-09:
# Problem

Visualization of hydroelastic contact surfaces in `drake_visualizer` is *painfully* slow. While part of this may be attributed to compactness of LCM messages or inefficient mesh topologies (e.g., duplicate vertices), the primary cause comes down to how the mesh is expressed to VTK: as a ]bunch of individual polygons](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py#L681-L684).

# Diagnoses

By registering each mesh triangle as its own VTK entity, the rendering of the mesh becomes CPU bound instead of GPU bound. More meshes are more VTK objects where each is processed in a large generic fashion. Instead, of all vertex and triangle were provided as a single mesh, it would be a single object and the GPU would be doing all the heavy lifting.

# Solution

Use the data in the LCM message to populate the polydata of a single polygonal object.

DamrongGuoy, 2021-01-14:
On my Puget, this example

 `bazel run //geometry/profiling:contact_surface_rigid_bowl_soft_ball`

was captured in these three movies to show how contact visualization performs right now.  In the first movie with both contact pressure and contact mesh, I saw only one frame for the entire 10s sim.  In the second movie with only contact mesh no pressure, the movie has more frames to show.  The third movie hides the contact surface and shows the real time factor about 1.0X.

https://user-images.githubusercontent.com/42557859/104661600-b7d4a800-567d-11eb-8f9b-49a58023f9b2.mp4

https://user-images.githubusercontent.com/42557859/104661604-ba370200-567d-11eb-906f-d1760495a43a.mp4

https://user-images.githubusercontent.com/42557859/104661610-bc995c00-567d-11eb-9bcc-6f9d81423da2.mp4

DamrongGuoy, 2021-01-14:
Thanks to @xuchenhan-tri for planning to work on it.

SeanCurtis-TRI, 2021-02-11:
Resolved by #14622

closed 2021-02-11 by SeanCurtis-TRI