RobotLocomotion/drake #13670: mkdoc: Swap libclang backend out with pygccxml

EricCousineau-TRI, 2020-07-11:
As part of #7889, Kitware has shown me a pretty neat demo setup with [pygccxml](https://pygccxml.readthedocs.io/en/master/index.html) (using CastXML).
My initial concern with CastXML was that we'd have to write a new adapter on top of the XML, but `pygccxml` (of course) takes care of that, and it's API looks rather comprehensive.

Given this, I'd like to see how difficult it would be to swap out `libclang` with CastXML.
The goal is to see how much we can gain from `pygccxml` in terms of reducing dumb parsing edge cases from `libclang`, as well as see how much (if any) non-negligible overhead we get from using XML as an intermediate representation.

My target is to try this in the next 2 weeks. (Of course, that may slip ;)

\cc @jamiesnape @josephsnyder

EricCousineau-TRI, 2020-07-11:
Filed initial wip in #13671

jwnimmer-tri, 2021-11-10:
@EricCousineau-TRI can you offer any perspective on what benefit pygccxml might offer us today (if we switch to it for mkdoc)?  Our docstring generation isn't particularly speedy or easy to modify, but neither does it seem to have gaping holes or missing strings (#12075 is the only counter-example I know of).  I am happy to still entertain this area for exploration, but I'd like to know what kind of outcomes we might be looking for.

closed 2024-03-12 by jwnimmer-tri