RobotLocomotion/drake #13672: MultibodyPlant doesn't document its default collision filtering behavior

SeanCurtis-TRI, 2020-07-13:
# Problem

A [recent question came up](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1594404370263800): 

> If I have two bodies A and B connected by a prismatic joint, are collisions between geometries on A and B ignored by default?

The [documentation on collision filtering in SceneGraph](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_scene_graph.html) discusses the *mechanisms* for introducing filters on collisions, but doesn't provide any insight into this particular question.

# Proposal

As outlined in the slack thread (linked above), `SceneGraph` shouldn't know about "adjacent bodies". And, in reality, it can't. `MultibodyPlant` presents the bodies as a flat array of frames, all posed relative to the world frame. So, it would be impossible for `SceneGraph` to make any decisions about the geometries based on opaque relationships between those frames. That is clearly the province of `MultibodyPlant`. 

- Confirm that `MultibodyPlant` contact is clearly documented as precluding contact between geometries on "adjacent" bodies.
- Make sure it's reasonably discoverable (this *might* include a cross-reference in the `SceneGraph` collision filtering documentation as "examples" of the API.)

rpoyner-tri, 2022-01-25:
Perhaps adding some text to `multibody/plant/contact_model_doxygen.h` is the right way to resolve this.

rpoyner-tri, 2022-02-01:
Having looked at contact_model_doxygen, I am shying away from it for now. It is mostly point-contact-era theory, AFAICT, and may want a general revisiting. I think for now, I'll just add some doc to the description of Finalize(), which is where the magic collision filters are created.

PR #16490 (2022-02-03): [multibody] Document default collision filtering
Closes: #13672
Relevant to: #16422

Write down what we do now, add a clarification and cross-reference to the
SceneGraph docs, and delete some obsolete text.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16490)
commit 294e1bc3 (2022-02-03): [multibody] Document default collision filtering (#16490)

* [multibody] Document default collision filtering

Closes: #13672
Relevant to: #16422

Write down what we do now, add a clarification and cross-reference to the
SceneGraph docs, and delete some obsolete text.
closed 2022-02-03 by rpoyner-tri