RobotLocomotion/drake #13673: [geometry] Runtime support for controlling what geometry is visualized

SeanCurtis-TRI, 2020-07-13:
# Problem

It has been a common request (in [stackoverflow](https://stackoverflow.com/questions/62777719/view-collision-models-in-drake/62778146?noredirect=1#comment111123820_62778146) as well as in person) that Drake offer a more agile approach to visualizing geometries.

Currently, it is possible to [control what geometry (selected by __`Role`__) is visualized at the `Diagram`-building time](https://github.com/RobotLocomotion/drake/blob/master/geometry/geometry_visualization.h#L79-L80). However, in practice it is convenient to be able to toggle back and forth between visualization geometry and collision geometry. This build-time API precludes the possibility of this dynamic, run-time workflow.

# Design considerations

- Addressing this issue *must* be done on a per-visualizer basis. There is no general solution inside `SceneGraph` that will resolve all visualizers; each one uses different mechanisms for loading and identifying geometries.
- It isn't clear what level of control should be offered. Coarse granularity will lead to a simpler implementation, but fine granularity, assuming a good UX design on the visualizer side, might be more convenient.
    - Toggle the whole world?
    - Toggle single "frame" (which corresponds to `MultibodyPlant` bodies).
    - Toggle selection of "frames".

# Proposal

1. Use `drake_visualizer` as the initial basis for working through resolving this issue. Defer implementing parallel functionality in other visualizers until we have a vetted and working UX.
2. Implement only the coarsest level of granularity (toggle the whole scene between `proximity` and `illustration` geometry.
3. Implementation
   - The visualizer will have a third visualization option: `proximity`, `illustration`, `both`.
   - LCM message will be augmented (in some way) to indicate distinction between `proximity` and `collision` geometry.
   - Introduce new `drake_visualizer` GUI element (menu option) that toggles the whole world (provided via a python plug-in).

Note: The assumption is by implementing the above today, we enable future workflows with finer granular control.

jwnimmer-tri, 2022-05-25:
As of today, I think a good target would be to use the `lcmt_viewer` messages for this (i.e., by having the diagram transmit the proximity geometry shapes and poses as well) but to target `meldis` as the UI to experiment with user controls.

I imagine that a slider for "proximity alpha" (0-100%) might be a sufficient UI.  The display already has a scene tree with checkboxes to turn models on/off.

jwnimmer-tri, 2022-06-01:
Also FYI -- one option here for the `DrakeVisualizer` system would be to use the same message types for the proximity-geometry messages as the visual-geometry message, but to send them on a different LCM channel name.

rpoyner-tri, 2022-06-01:
I have a rough running prototype that multiplexes on robot_num. I'm not entirely sold on it yet.

I will consider the channel name multiplex scheme.

Don't all of these tricks evaporate when we transition away from drake visualizer?

jwnimmer-tri, 2022-06-01:
I don't foresee `meldis` disappearing for a long time.  Having a permanently-running visualization display for ongoing, one-process-at-a-time simulations is a huge win.  Launching a `MeshcatVisualizer` from a standalone simulation program is awkward to restart every time.  Maybe if #17085 was different it would be better?

jwnimmer-tri, 2022-07-07:
If the user sets alpha all the way down to zero, we could notice that and hide the geometries entirely.

If the problem still happens at non-zero alphas though, we'll probably want to keep digging.  (In my mind, a common use case here will be having one or both alpha-blended most of the time, so neither one fully obscures the other.)

rpoyner-tri, 2022-07-21:
The UI-enabling patch is near completion. @jwnimmer-tri has some ideas for sugar APIs; passing this issue over to him.

jwnimmer-tri, 2022-07-25:
Other tasks to consider before closing this issue.  (Some of these might turn into follow-up issues.)

- [ ] Add setup / configuration sugar
   - I have a sugar PR under review in Anzu.  We'll work on bringing that into Drake soonish.
- [ ] Port all sample programs (and tutorials) to new sugar
- [ ] MeshcatVisualizer natively supports the same new feature as meldis (alpha sliders, on/off geometry check box in the scene tree).
- [ ] Update geometry_inspector / show_model to parity.
- [ ] Move geometry_inspector / show_model into pydrake as installed library code and rework MbP tutorial to use them.

jwnimmer-tri, 2022-08-03:
I _think_ that all of the unfinished work has been filed as either pull requests or issues now (linked immediately above).  As such, we can close the overall meta-issue.

closed 2022-08-03 by jwnimmer-tri