RobotLocomotion/drake #13678: Initializing a MultibodyPlant segfaults when time_step=0

deq2, 2020-07-14:
I'm trying to use direct collocation with a custom urdf robot. To do this I need to initialize the the MultibodyPlant with time_step=0.0. However, the program fails with a segfault. I've added a code snippit below:
https://gist.github.com/deq2/b39bf8d1f8a1108de000ead2ac90f0f6

rpoyner-tri, 2020-07-14:
I've adapted the snippet to live in the drake tree in my branch; seg-fault reproduced. Investigating.

sherm1, 2020-07-14:
Looks like it declares the MultibodyPlant to be continuous but then asks (at the end) for the discrete state variables (which don't exist in that case). I believe if it asked for continuous state variables all would be well.

rpoyner-tri, 2020-07-14:
@sherm1 Agreed. I'm pondering whether the API hazard with discrete state is sufficiently documented.

rpoyner-tri, 2020-07-14:
Ah. The code path that yields the seg-fault runs through a DRAKE_ASSERT that is apparently disarmed in libdrake.so as built. Do we provide an asserts-enabled version of libdrake.so with instructions, to help external users debug this sort of thing?

rpoyner-tri, 2020-07-14:
@deq2 this branch https://github.com/rpoyner-tri/drake/tree/gh13678 has a couple of commits showing the adapted code, and some changes I made to avoid the misuse and print out various bits of context info.

sherm1, 2020-07-14:
> The code path that yields the seg-fault runs through a DRAKE_ASSERT ... Do we provide an asserts-enabled version of libdrake.so ...

That should instead be a nice error message like "You asked for a reference to discrete variables but there aren't any in this model." Minor user misunderstandings shouldn't cause segfaults!

rpoyner-tri, 2020-07-14:
@sherm1 again, agreed. Presumably that's just a change of error handling. I'll look at the options.

deq2, 2020-07-14:
Ah I understand the issue. Thanks for getting back to me so quickly.

rpoyner-tri, 2020-07-14:
@deq2 glad we could help. I'll have a PR up at some point for an improved error message, but that shouldn't hold you up.

rpoyner-tri, 2020-07-14:
https://github.com/robotlocomotion/drake/pull/13681 is one proposal for a fix.

rpoyner-tri, 2020-07-17:
Patch merged.

closed 2020-07-17 by rpoyner-tri