RobotLocomotion/drake #13689: parsing: Allow role specification for //visual tags

EricCousineau-TRI, 2020-07-15:
From Anzu issue from @siyuanfeng-tri:

> I am able to simulate the full stack clutter example [...] on a dev branch (at 0.25 rt). One of the hacks I had to do is to disable the cameras' visual geometry in the sdf model. Otherwise, the grey boxes we added to represent the collision geometry of the camera would show up in the simulated camera feeds.
> 
> The concept has already by planned and supported by SG, https://github.com/RobotLocomotion/drake/blob/d8b1c179a5/geometry/dev/geometry_roles.h#L187. We can add custom tags in sdf to support role specification, or change them programatically in runtime.

@SeanCurtis-TRI has already implemented the runtime change in code.
However, it would be nice to at some point specify it in URDF/SDformat, using something like `<drake:geometry_role perception="false" illustration="true"/>`.

SeanCurtis-TRI, 2020-07-15:
One alternative is to make use of the `drake:accepting_renderer` tag. There are two options:

1. List a nonsense renderer name (one that doesn't exist in your simulation). This should work today.
2. We modify the [parsing semantic such that it accepts an empty list](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_urdf_geometry.cc#L389-L397).

This allows for perception = false, illustration = true. But doesn't allow for perception = true, illustration = false.

I'm loathe to inject the *specific* tag suggested because I think it'll be redundant of what we actually need as we flesh out illustration and perception properties that will be specified in URDF/SDF files. However, we *definitely* require finer granularity in specifying roles for geometry. Two tags (`<visual>` and `<collision>`) spread across three roles is dissatisfying. I can imagine we do something akin to `<drake:proximity_properties>` (inside `<collision>`) and have `<drake:illustration_properties>` and `<drake_perception_properties>` inside `<visual>`.

EricCousineau-TRI, 2020-07-15:
Aye, per-role properties sound good. I think the finer-granularity drake-specific properties make total sense. Is there a way to bake in the "this is enabled/disabled" into that tag? e.g.
* `//collision/drake:proximity_properties/@enabled`
* `//visual/drake:illustration_properties/@enabled`
* `//visual/drake:perception_properties/@enabled`?

(note: [XPath syntax](https://www.w3schools.com/xml/xpath_syntax.asp))

EricCousineau-TRI, 2020-07-15:
Also, relates: https://github.com/RobotLocomotion/drake/issues/11746

SeanCurtis-TRI, 2020-07-16:
I'm not entirely clear on what the semantics of `@enabled` would be. I can imagine a couple of things I'm not overly fond of and would like to see what you have in mind.

EricCousineau-TRI, 2020-07-16:
`@enabled = True` (the default) implies this role should be used, and will have information.

`@enabled = False` implies this role should not be used, and should have *zero* information.
Additionally, we may want to throw if all available roles for a tag are disabled.

So yeah, I can see where your lack of fondness comes from, but not sure of what the better solution would be, as the underlying problems (redundancy and consistency) would end rearing it's face in any form?

SeanCurtis-TRI, 2020-07-16:
Yep. You see it.

For `<collision>` it seems odd to declare a `<collision>` tag and then disable it using `drake:proximity_properties`. Although that may be perfectly serviceable. SDF/URDF don't particularly provide alternate methods for "disabling" geometry short of cutting it from the file. If we bought into that concept, then throwing wouldn't be what we would do.

It's all muddied up by `<visual>` vs `<collision>` and the ambiguous interpretation of `<visual>`. There's a *third* option. We add another tag `<drake:render>` as a sibling of `<visual>` and `<collision>`. That would disambiguate at the cost of requiring parsing magic to maintain backwards compatibility and making things even more verbose than currently.

If I had my druthers, the architecture would've been reversed. Instead of:

```xml
<link>
  <visual>
    <geometry ... />
  </visual>
  <collision>
    <geometry ... />
  </collision>
</link>
```

It would be this instead:

```xml
<link>
  <geometry ...>
    <visual .../>
    <collision ... />
  </geometry>
</link>
```

But that should be apparent about how I think of geometry.  :) In other words `<collision>` and `<visual>` are *useless* without `<geometry>`, but `<geometry>` without one of those is meaningful. The role a geometry plays is a property it has and not the other way around.

I think this will require some more consideration.

EricCousineau-TRI, 2020-07-16:
I agree with your reversal - I know you've mentioned it a few times, but yeah, would love to see it come to fruition.
Will see if I can at least briefly seed that discussion for SDFormat 1.9 (and as always, I think URDF should only ever be legacy, and all dev work on it should stop ;).

That being said, I'd still like the practical short-term solution.
I think doing some form of `@enabled=False` will work, and if you specify `collision/drake:proximity_properties/@enabled=False`, then you get an error saying "yo, comment that out". Same if you comment out all perception stuff.

I would vote against `drake:render` for now, as I think the proper short-term solution is add children to `visual` / `collision`, and the long term solution is what you suggested.

Thoughts?

SeanCurtis-TRI, 2020-07-16:
So, to summarize my understanding of the design:

1. We have three tags `<drake:proximity_properties>`, `<drake:illustration_properties>`, `<drake:perception_properties>`.
    - `<drake:proximity_properties>` already exists.
2. They would be child tags of the `<collision>`, `<visual>`, and `<visual>` tags, respectively.
3. Each has an `enabled` attribute.
4. Presence of a tag is required for the geometry to receive the corresponding role.
5. Setting `enabled` to false is equivalent to the tag not being present.
6. If a geometry is defined and has no "present" role tags, that is considered an error.

If that is correct:

I'm not entirely sold on the value of `enabled`. The only real value is to have illustration and perception tags both defined in the spec file, but optionally be able to turn one or the other on/off independently. That is just a tiny bit of sugar around simply cutting the text. Is it worth it?

jwnimmer-tri, 2022-03-09:
See also #16043.

jwnimmer-tri, 2022-03-09:
I think allowing for this in the SDFormat text is a nice win.

On another angle, maybe a nice way to support these use cases is when calling `Parser::AddModelFromFile` the user could choose whether `<visual>` geometry defaults into illustration, perception, or both.  That would allow using one model file for multiple purposes, without forking it into different copies.  Or alternatively, make it a one-liner to change their mind after the fact (e.g., strip off perception properties) ala #16754.

jwnimmer-tri, 2023-04-11:
I'll drop another (somewhat related, somewhat different) feature request here since it might play a part in designing the solution.

Right now, if the user defines a geometry in SDFormat they use `<visual>` and `<collision>` and **very frequently** both are the same geometry, to the point where people often use xacro to copy-paste visual and collision identically from the same definition.  When loaded into MultibodyPlant, this ends up with two copies of the geometry (with different IDs).  If the the user wants to tweak the shape (e.g., change the radius of a sphere) they need to do it on both copies.  It would be convenient if they could do it just once.

That would mean either that the parser recognizes and de-duplicates the two stanzas to use the same geometry ID, or we could imagine allowing `<drake:illustration_properties>` attribute on `<collision>` geometry, to add the illustration role onto the collision shape.

jwnimmer-tri, 2023-05-10:
Upthread, there have been proposals to specify perception-only geometry using something like this syntax:

```xml
<link>
  <collision>
    <!--- This shape would only be used as proximity geometry. -->
    <geometry ... />
    <drake:proximity_properties ... />
  </collision>
  <visual>
    <!--- This shape would be used as both illustration and perception geometry. -->
    <geometry ... />
  </visual>
  <visual>
    <!--- This shape would only be used as illustration geometry. -->
    <geometry ... />
    <drake:illustration_properties ... />
  </visual>
  <visual>
    <!--- This shape would only be used as perception geometry. -->
    <geometry ... />
    <drake:perception_properties ... />
  </visual>
</link>
```

When any kind of property element is present on a `<visual>`, we would assign only those specified properties.  Only when neither flavor of properties are given on a `<visual>` would we apply the current default of adding both illustration and perception roles to those geometries.

The key objection to this proposal is that if the user were to load the file in a different simulator, they would end up with (presumably) two copies of their illustration geometry, leading to z-fighting and general confusion.

I can only think of two ways around this:

(1) When providing bonus perception-only geometry for use by Drake, advise the user to set `<transparancy>0</transparancy>` on it so that other simulators will not display it.  Drake would need to promise some hack to disregard this option when rendering (e.g., it was set to zero then ignore it for the perception role).

(2) Use `<drake:visual>` or `<drake:perception>` instead of `<visual>` for perception-only geometry.  It would have an identical schema as `<visual>`, but in our own XML namespace.  Other simulators would ignore it.

jwnimmer-tri, 2023-05-15:
My vote is for *2b, because I think it leads to the simplest rules to teach users:
- When geometry should be both roles, declare it as normal in SDFormat.
- When geometry should be illustration-only in Drake but visual to other simulators, add the `drake:illustration` property to the existing SDFormat `<visual>` element.
- When geometry should be perception-only in Drake and ignored by other simulators, use `drake:perception` to declare it.

The use case it does not support is adding illustration geometry that is specific to Drake but ignored by other simulators.  We don't seem to have a need for that.

jwnimmer-tri, 2024-02-19:
As we work towards optimizing various geometry consumes with different "level of detail" mesh assets, that might be an important motivation for this model file syntax issue.

jwnimmer-tri, 2024-10-02:
See also http://sdformat.org/spec?ver=1.11&elem=visual#visual_visibility_flags.  That could be another way (or _the_ way) to control visibility.

PR #22013 (2024-11-20): Introduce sdformat semantics for Drake's visual roles
This introduces a working example of an sdf file.

Closes: #13689

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22013)
closed 2024-11-20 by rpoyner-tri