RobotLocomotion/drake #13692: parsing, urdf: See `SetDefaultFreeBodyPose()`

EricCousineau-TRI, 2020-07-16:
This is a speculative feature to enable the following prototype:
https://github.com/RobotLocomotion/drake/pull/13128#pullrequestreview-449919203
Relates #13691 

\cc @sammy-tri

sammy-tri, 2020-07-16:
I think this might be difficult to do in practice.  See my comments about the different cases of parent bodies in #13691.

jwnimmer-tri, 2022-05-02:
@EricCousineau-TRI the issue overview is somewhat terse.  Could you expand upon the feature request to summarize, please?  What specifically is being requested, and why it is relevant/important to make this change?

EricCousineau-TRI, 2022-05-03:
Upon rereading the references issues / PR, I think this issue is redundant in terms of #13691 - more or less, this is saying "to properly support an initial pose, we should use `SetDefaultFreeBodyPose()` when parsing via URDF, to ensure reposturing when including w/ SDFormat works correctly"

Closing - thanks for the ping!

closed 2022-05-03 by EricCousineau-TRI