RobotLocomotion/drake #13700: Headers including OpenGL headers are being installed

jamiesnape, 2020-07-17:
[`geometry/render/gl_renderer/opengl_includes.h`](https://github.com/RobotLocomotion/drake/blob/master/geometry/render/gl_renderer/opengl_includes.h) is the direct include.

CC: @jwnimmer-tri @SeanCurtis-TRI

jamiesnape, 2020-07-17:
The question is, what is the API of `gl_renderer`? Which headers?

jwnimmer-tri, 2020-07-17:
Yes, exactly.  The factory is the only public header.  Everything else is (or should be) only using the `internal` namespace.

@SeanCurtis-TRI Note that `render_engine_gl.h` should move into the internal namespace; we missed that during the original reviews.  It is not a public class.

jamiesnape, 2020-07-17:
> Note that render_engine_gl.h should move into the internal namespace; we missed that during the original reviews. It is not a public class.

That explains my confusion. Thanks.

jamiesnape, 2020-07-17:
Maybe I am missing something, but the visibility does not seem quite right, even for non-install tree users. I would expect `public` on a library with just `render_engine_gl_factory.h` and (depending on how you want to have the utility classes available) `private` on everything else?

jwnimmer-tri, 2020-07-17:
The package level `default_visibility = ["//geometry/render/gl_renderer:__subpackages__"]` is admittedly a bit silly now that there is no `dev` sub-folder.  That's functionally equivalent to private, so we should just respell it as private.

The `drake_cc_package_library...` is public visibility, and that's what people should deps on.

Sean and I did discuss carving out a `drake_cc_library(name = "render_engine_gl_factory")` to permit a user to refer to that explicitly (so as to match the include path more precisely), but decided there wasn't any real value there.  If all call sites uniformly just refer to the package library, that is succinct and uniform.

jamiesnape, 2020-07-17:
I see, `drake_cc_package_library` hides its visibility declaration in its implementation. There is so much `package(default_visibility = ["//visibility:public"])` going on here that I never even noticed that. I guess it makes functional sense, but it reads very badly when you have a global `default_visibility` declaration that is not public.

jwnimmer-tri, 2020-07-17:
Yup, that tripped me up earlier this week as well.  I think we should probably change to require explicit visibility on those, with a suitable advice / fixer to help users with the boilerplate.

SeanCurtis-TRI, 2020-07-17:
Ok, I'll work up a PR.

PR #13807 (2020-08-04): gl_renderer: Do not install private headers
Split the renderer out from the factory, for clarity.

Closes #13700.
Closes #13702.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13807)
closed 2020-08-04 by jwnimmer-tri