RobotLocomotion/drake #13718: Feature request: Inverse dynamics controller to support under-actuated systems

huihuaTRI, 2020-07-21:
## Issue
The [InverseDynamicsController](https://github.com/RobotLocomotion/drake/blob/b896a921508ef55ed1b83368bb2596b4017fec70/systems/controllers/inverse_dynamics_controller.h#L50) currently only supports fully actuated systems, which is limiting when work with systems with a floating base or robot that has free moving joints in the middle of the tree.

For the systems with a floating base but the upper body is fully actuated, the work-around is to create a separate controller plant with the base welded to the ground. While this is not straightforward, it does not require too much work.

However, for the systems that have a free joint in the middle of the tree, there is no elegant way to use this controller. The cases of **"in the middle of the tree"** are opposed to the floating base cases, in which the base is at the root of the tree. 

An example of this case is a robot that has a tool rigidly attached to the hand. The tool (such as a [swiffer](https://www.amazon.com/Swiffer-Sweeper-Starter-Cleaning-Products/dp/B07YQDH1N1/ref=asc_df_B07YQDH1N1/?tag=hyprod-20&linkCode=df0&hvadid=416693387988&hvpos=&hvnetw=g&hvrand=4777983949884981142&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9060440&hvtargid=pla-890136224371&psc=1&language=en_US&tag=&ref=&adgrpid=97671770327&hvpone=&hvptwo=&hvadid=416693387988&hvpos=&hvnetw=g&hvrand=4777983949884981142&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9060440&hvtargid=pla-890136224371)) itself will have unactuated joints. In this case, you would need a different version of urdf, in which the free joints are fixed. This urdf is then used for the controller plant. Keeping two versions of the same robot modeling file generally is not a favorable solution. They introduce confusion and can easily get out of sync.

## Proposition
This thread is to start the conversation to generalize the InverseDynamicsController so that it can support under-actuated systems. As the first step, we don't consider joint limit yet.

The blocker right now is the [InverseDynamics](https://github.com/RobotLocomotion/drake/blob/b896a921508ef55ed1b83368bb2596b4017fec70/systems/controllers/inverse_dynamics.h#L33) system does not consider applied external spatial forces, which depends on the actuator torque. It would end up with an algebra loop. One idea is to use the external spatial forces calculated in the previous timestep to approximate the current forces. See this [reference](https://github.com/RobotLocomotion/drake/files/4950711/nakanishi_icra2007.pdf).

I am just tagging multiple people here to solicit ideas and suggestions. Thank you in advance!
@siyuanfeng-tri @amcastro-tri @sammy-tri @sherm1 @rcory @EricCousineau-TRI

amcastro-tri, 2020-07-21:
> the InverseDynamics system does not consider applied external spatial forces, which depends on the actuator torque

Could you be very specific about that? example? the actual physics being modeled?
I didn't read the paper, but if you end up with an algebraic loop I'd say there is no work around. You'd either need to reformulate the math or probably come to the conclusion that  InverseDynamics is not the right tool.

Summarizing, I'd like to understand the math first without digging at all into the Drake implementation details. I don't want to think about algebraic loops at all. Does that paper reference implement the exact math you want? (trying to figure out if to spend time reading it)

siyuanfeng-tri, 2020-07-21:
there was a qp inverse dynamics controller i wrote many years ago for humanoid, which might be more relevant? but that was nuked because of deprecation and no real demand.

huihuaTRI, 2020-07-21:
@amcastro-tri Section 2 and Equation 5 in the reference should be enough to show you what I am talking about. I would write the same equations to show you a math example.

@siyuanfeng-tri Is it possible to find the code somewhere?  Pure gravity compensation may also be an option for me. I figure the performance would not be as good as the inverse dynamics controller.

siyuanfeng-tri, 2020-07-21:
https://github.com/RobotLocomotion/drake/pull/9855/files#diff-40a44bb5c7900b0e4ceb08c545f61389 is a starting point.. i can't find it right now, it would be something in attic/systems/controllers/qp_inverse_dynamics/stuff.

huihuaTRI, 2020-07-21:
Thanks @siyuanfeng-tri!  For people who are interested, here is the commit: 7d5c4736e91e2d64d5af2980f4dca77cd3c9b15c before this controller got removed.

RussTedrake, 2020-07-24:
Drive-by comment: the issue title generates a “parse error” in my mind.  inverse dynamics is ill-posed for underactuated systems.  (you cannot find a torque to produce every acceleration; you essentially have written “invert a rank-deficient matrix”).  I believe that what you mean is that you only want to control a subset of the configuration variables — ones with collocated actuators.

The QP controller does this, of course.  But it also does more than you need.  It’s probably only a QP if you want to reason about friction cones.

huihuaTRI, 2020-07-26:
Thanks for making it clear. Yes, I do mean calculating the control effort for the actuated joints. Normally, the users have to load a separate plant for this purpose if the system is under-actuated (e.g. floating base). I am wondering whether it's possible to let the system `InverseDynamicsController` figure it out internally. Therefore, the users won't need to do the extra steps anymore.

sherm1, 2022-04-26:
We probably still need something like this for mobile manipulators but I'll close it now since it has been a few years. I expect it will come up again with a new application.

closed 2022-04-26 by sherm1