RobotLocomotion/drake #13726: mkdoc.py code moved from third_party

jamiesnape, 2020-07-21:
#13651 split up `third_party/com_github_pybind_pybind11/mkdoc.py`, which is a good thing. Unfortunately, it relocated some of its (third-party) code to `tools/workspace/pybind11/mkdoc_comment.py`, i.e., from the `third_party` directory to a regularly licensed directory. The code has been heavily modified, but it does retain copyright "2016 Wenzel Jakob" for part of it. I guess that this stage, it probably just needs a suitable statement in the file header, but code moving to/from `third_party` probably should be flagged in review in general.

FYI @BetsyMcPhail @EricCousineau-TRI @jwnimmer-tri.

jamiesnape, 2020-07-21:
Aside: The `third_party` directory is a losing battle. My packaging tools are telling me that there are more third-party files outside of `third_party` than within it.

EricCousineau-TRI, 2020-07-21:
Putting under `pydrake` for now and on that board. Seems like an easy 5-10min fix? (just for `mkdoc` ofc)

PR #13769 (2020-07-30): Move `mkdoc.py` from `third_party`
Relocate `mkdoc.py` to the same directory as `mkdoc_comment.py`. Remove the
LICENSE file and (now empty) `third_party/com_github_pybind_pybind11`
directory. Add copyright information to the two files that are derived
from original `mkdoc.py` in pybind11.

Resolves #13726

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13769)
commit 2a9581a4 (2020-07-31): Move `mkdoc.py` from `third_party`

Relocate `mkdoc.py` to the same directory as `mkdoc_comment.py`. Remove the
LICENSE file and (now empty) `third_party/com_github_pybind_pybind11`
directory. Add copyright information to the two files that are derived
from original `mkdoc.py` in pybind11.

Resolves #13726
closed 2020-07-30 by BetsyMcPhail