RobotLocomotion/drake #13736: [geometry] Non-determinism in contact results

SeanCurtis-TRI, 2020-07-22:
The various contact results queries have a contract that is *too* sloppy to guarantee determinism.

1. Contact between two geometries A and B is guaranteed to be *reported* as (A, B). But we need to make sure that it is computed in a consistent manner. 
    - Contact between two geometries consists of transforming them to a common frame. Typically one geometry into the other geometry's frame. The choice of which geometry is transformed can affect the final answer based on round-off error introduced in the transformation process. While we won't lay claims that transforming one way or the other is more *correct*, we need to be consistent that given two geometries one will *always* be transformed to the other so that the rounding error is the *same*.
2. The aggregate contact results should be consistently ordered in a similar way. IF pairs (A, B), (C, D), (E, F) all track contact, they should be "well ordered". Because floating-point arithmetic of commutative math is only quasi-commutative, guaranteeing this order is required for guaranteed determinism in the resulting math.

These issues are probably all resolved when this gets resolved; we'll defer that judgment until we observe the resolution of *this* PR in the wild.
related to #13505, #13506, #13507

SeanCurtis-TRI, 2020-07-22:
cc @rpoyner-tri, @EricCousineau-TRI

rpoyner-tri, 2020-07-22:
This issue looks to me like a more specific version of #13507 . How do you feel about closing that one in favor of this one?

SeanCurtis-TRI, 2020-07-22:
Suits me.

SeanCurtis-TRI, 2020-07-22:
Assuming this really is the full contribution of SceneGraph to non-determinacy. But I'm willing to go out on a limb and say we can re-open that if we're proven wrong.

DamrongGuoy, 2020-07-23:
What should be the API contract to guarantee determinism? Is this the right contract?
**Contact(g0,g1) and Contact(g1,g0) are consistent to the last bit**

For example, if we have:
```C++
drake::geometry::PenetrationAsPointPair pair_g0_g1;
drake::geometry::PenetrationAsPointPair pair_g1_g0;
```
We would want this function to return true (I'm not sure whether this is a good function name):
```C++
EXPECT_TRUE(EqualWhenSwitchOrderPair(pair_g0_g1, pair_g1_g0));

bool EqualWhenSwitchOrderPair(const PenetrationAsPointPair& g0_g1, const PenetrationAsPointPair& g1_g0) {
  return g0_g1.id_A == g1_g0.id_B &&
         g0_g1.id_B == g1_g0.id_A &&
         g0_g1.p_WCa == g1_g0.p_WCb &&
         g0_g1.p_WCb == g1_g0.p_WCa &&
         g0_g1.depth == g1_g0.depth;
}
```

In addition to `PenetrationAsPointPair`, this will include `SignedDistancePair`like this:
```C++
bool EqualWhenSwitchOrderPair(const SignedDistancePair& g0_g1, const SignedDistancePair& g1_g0) {
  return g0_g1.id_A == g1_g0.id_B &&
         g0_g1.id_B == g1_g0.id_A &&
         g0_g1.p_ACa == g1_g0.p_BCb &&
         g0_g1.p_BCb == g1_g0.p_ACa &&
         g0_g1.distance == g1_g0.distance &&
         g0_g1.nhat_BA_W == -g1_g0.nhat_BA_W &&  // Need opposite sign when compare this vector.
         g0_g1.is_nhat_BA_W_unique == g1_g0.is_nhat_BA_W_unique;
}
```
Notice the opposite sign of the directional vector `g0_g1.nhat_BA_W == -g1_g0.nhat_BA_W`.

SeanCurtis-TRI, 2020-07-23:
While I'm not entirely sure what you've written, I believe that may be because you and I have different models in our minds. So, let me try to clarify. Please let me know if this alleviates your concerns.

If geometries g0 and g1 contact, then one of the invariants is that you can only *ever* get `id_A = g0` and `id_B = g1`. You can never, ever get `id_A = g1` and `id_B = g2`. Thus, for the contact there is a single unique representation of the contact -- not two equivalent representations. That's why the test is on the callback. If I call the callback as (g0, g1) or (g1, g0), I'm guaranteed that the result I get is `id_A = g0` and `id_B = g1`, with all other fields identical to the last bit.

Humorously, the idea that the result will always be reported with `id_A = g0` and `id_B = g1` has been part of the API since the beginning. However, it didn't guarantee that every bit would be exactly identical. (Now it does.) And then it takes it a step further in that a set of contacts will be consistently ordered.

However, you raise a *very* valid point that the same principle needs to be applied to the signed distance.

PR #13737 (2020-07-23): [geometry] Making contact results deterministic
This encompasses two things:
  1. Guarantees that if objects A and B make contact, that the result is identical regardless of which is "first" or "second".
  2. For multiple contacts, the results will be ordered in a reliable ordering (for a fixed set of poses).

This impacts the following queries documented in QueryObject:
  - ComputePointPairPenetration()
  - ComputeContactSurfaces()
  - ComputeContactSurfacesWithFallback()
  - FindCollisionCandidates()

resolves #13736

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion [... cut]
closed 2020-07-23 by sammy-tri