RobotLocomotion/drake #13738: mobilizers code uses doxygen-style comments (///) but is in an internal namespace

RussTedrake, 2020-07-23:
mobilizers are (almost) invisible to doxygen/sphynx.  but local reading of the code would suggest otherwise.

this is true of at least mobilizer.h and mobilizer-impl.h, and presumably their descendants.

Related to #11229.

mpetersen94, 2020-08-12:
Is the solution here to convert all the doxygen-style comments (///) to standard comments (/* */) or is there more to the solution as #11229 suggests is needed?

SeanCurtis-TRI, 2020-08-12:
`///` --> `//` is one option.
If you prefer the `/** */` over `///` then moving to the *single star* `/*  */` is also a valid choice.  

(Note `///` and `/** */` are reserved for comments that should get rendered for doxygen.)

sherm1, 2020-08-12:
@mpetersen94: we have Doxygen configured to ignore things that are in an `internal::` namespace (like Mobilizer). So the problem here is just that the comments suggest something that isn't true to someone reading the code. (That this code is part of the user-facing API.)

mitiguy, 2020-12-15:
Related to this issue, change comments in multibody_tree.h from /// to //.

mitiguy, 2020-12-21:
I think PR #14459, PR #14461, PR #14468, and PR #14474 resolve most (or all) of this issue within the headers files in the multibody and math subdirectories.

closed 2020-12-21 by mitiguy