RobotLocomotion/drake #13754: Missing API -- Frame::CalcSpatialAccelerationFromVdot() and/or Frame::CalcSpatialAcceleration()

mitiguy, 2020-07-28:
In frame.h, it would be helpful for a method such as the following to exist:
SpatialAcceleration<T> Frame::CalcSpatialAccelerationFromVdot(context, frame_A, p_AoAp_A, frame_E, const VectorX<T>& known_vdot);

Related: In frame.h, there are already methods for pose and spatial velocity such as:
SpatialVelocity<T>  CalcSpatialVelocityInWorld(context).
SpatialVelocity<T>  CalcSpatialVelocity(context, frame_M, frame_E);

Related: In multibody_plant.h, there is a method with a similar API:
SpatialAcceleration MultibodyPlant::CalcBiasSpatialAcceleration(context_, JacobianWrtVariable::kV,  frame_A, p_AoAp_A, frame_W, frame_W);

Related: In multibody_plant.h, there are somewhat related heavyweight methods, e.g.,
void CalcSpatialAccelerationsFromVdot(context, known_vdot, bool ignore_velocities,  std::vector<SpatialAcceleration<T>>* A_WB_array);

Related: In multibody_plant_jacobians_test.cc, for the Kuka robot, I am trying to create a more sophisticated test for the method MultibodyPlant::CalsBiasSpatialAcceleration() with a non-world measured-in frame.  Testing would be easier if the proposed Frame::CalcSpatialAccelerationsFromVdot() existed.

sherm1, 2020-07-28:
The missing API in the family with `CalcPose()` and `CalcSpatialVelocity()` is `CalcSpatialAcceleration()`, not `CalcSpatialAccelerationFromVdot()` which would be more of an internal or at least Advanced method.

samzapo, 2021-11-30:
I have need of this requested function, where I might use `frame_Fa.CalcSpatialAcceleration(context, frame_Mb, body_frame_E)` for getting a spatial quantity `Vdot_MbFa_E`.

Where either frames `Fa` and `Mb` might be a `FixedOffsetFrame`.

 If frames `F` and `M` are floating body frames (i.e., they're in motion), there there are a number of calculations that would be nice to have in a convenience function.

sherm1, 2021-12-01:
cc @mitiguy

amcastro-tri, 2021-12-09:
@samzapo, until this lands, notice we have the API `MultibodyPlant::get_body_spatial_accelerations_output_port()` that puts per body spatial accelerations in an output port.

mitiguy, 2024-06-03:
PRs https://github.com/robotlocomotion/drake/pull/13773 [Body::EvalSpatialAccelerationInWorld(), Frame::CalcSpatialAccelerationInWorld()] and https://github.com/RobotLocomotion/drake/pull/16760 [Frame::CalcSpatialAcceleration(measured_in_frame, expressed_in_frame)] and https://github.com/RobotLocomotion/drake/pull/17117 [Frame::CalcRelativeSpatialAcceleration(); Frame::CalcRelativeSpatialAccelerationInWorld()]
resolve issue https://github.com/RobotLocomotion/drake/issues/13754

closed 2024-06-03 by mitiguy