RobotLocomotion/drake #13771: spdlog warning under NDEBUG

jwnimmer-tri, 2020-07-30:
A warning reported by TRI's downstream CI:
```
INFO: From Compiling external/drake/bindings/pydrake/common/text_logging_pybind.cc:
In file included from external/drake/bindings/pydrake/common/text_logging_pybind.cc:12:0:
bazel-out/k8-opt/bin/external/drake/tools/install/libdrake/_virtual_includes/drake_shared_library/drake/common/text_logging.h:39:0: warning: "SPDLOG_ACTIVE_LEVEL" redefined
 #define SPDLOG_ACTIVE_LEVEL SPDLOG_LEVEL_TRACE
 
In file included from external/spdlog/include/spdlog/sinks/base_sink.h:12:0,
                 from external/drake/bindings/pydrake/common/text_logging_pybind.cc:7:
external/spdlog/include/spdlog/common.h:134:0: note: this is the location of the previous definition
 #define SPDLOG_ACTIVE_LEVEL SPDLOG_LEVEL_INFO
```

This is with assertions (NDEBUG) enabled.

Best guess, include statements are in the wrong order.

jwnimmer-tri, 2020-07-30:
Also visible in CI e.g. https://drake-jenkins.csail.mit.edu/view/Production/job/linux-bionic-clang-bazel-nightly-debug/619/consoleText

jwnimmer-tri, 2020-07-31:
@BetsyMcPhail do you have a moment to work on this issue, please?

We'd like to remove this warning from the build's output.

My guess is that this is the relevant section:
https://github.com/RobotLocomotion/drake/blob/dc2a9394d4f3800ae32740229bc8b891a6e4ee6c/bindings/pydrake/common/text_logging_pybind.cc#L2-L12

To remove the warning, I think we need `text_logging.h` to be included _before_ the spdlog headers such as `base_sink.h`.

If we just move it higher in the file, I think `clang-format-includes` might try to move it back down again.  I think if we surround the include with `/* clang-format off */` + `/* clang-format-on */` that should do it.  Alternatively, moving the `text_logging.h` include to the header file here instead of cc file would also do it.  Either way, we should have a short comment explaining the (important) placement to avoid this warning.

WDYT?

BetsyMcPhail, 2020-07-31:
Thanks for the detailed report - I was able to reproduce on my system. I went with changing the include order in source file.

PR #13781 (2020-08-01): Remove spdlog warning
Change the order of include statements to avoid the SPLOG_ACTIVE_LEVEL
redefined warning.

Resolves #13771

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13781)
commit a0a3e268 (2020-08-01): Remove spdlog warning (#13781)

Change the order of include statements to avoid the SPDLOG_ACTIVE_LEVEL
redefined warning.

Resolves #13771
closed 2020-08-01 by jwnimmer-tri