RobotLocomotion/drake #13776: All visualization systems should stop using initialization events to bootstrap drawing?

EricCousineau-TRI, 2020-07-30:
Per f2f, @jwnimmer-tri mentioned that using initialization events for visualizers to send their "bootstrap" messages (e.g. loading a robot or publishing markers) is a workflow that kinda makes things more complicated than necessary - see #10015 (issue), #13760 (pr) for examples.

One solution is to have visualization systems know when they need to send these messages themselves (e.g. the message hasn't been sent yet).

That sounds feasible; the only question is how systems should "know".

Options that I can think of:
- Just store undeclared state (e.g. mutable member variable), and the system publish once in its "lifetime".
- Somehow stash state in the context, but avoiding the need for discrete updates, etc. (e.g. cache?).
- Just take a reference to the `Context*` in question, and remember this for each context. May grow in an unbounded fashion, but :shrug: 

Ideally, I'd like to try this out on the little Python RViz shindig, then hoist that to Drake-ish stuff afterwards:
https://github.com/EricCousineau-TRI/repro/tree/5fabd7fc7eac064f09a6b2a82a715d9bd81e0d09/ros/drake_ros1_hacks

Thoughts?

\cc @sherm1 @jwnimmer-tri @rpoyner-tri @SeanCurtis-TRI

SeanCurtis-TRI, 2020-07-30:
Your issue is implying a problem but I can't infer it from the linked reading materials (or alternatively, I may attempt to infer several things, none of which is what you want). So, could you clearly summarize the problem (for which the links are merely *background* reading?  I'm going to hold off on guessing what that summary to avoid injecting noise into the discussion.

But I do have some thoughts on what you *have* written.

1. Undeclared mutable state and cache entries in caches are right out.
   - Both of these would fail if it needed to rebroadcast.
   - Using a cache entry for behavior would break if caching were disabled (which is anti-caching).
2. I'm not sure what you mean by "take a reference to the Context* in question. Can you elaborate?

Ok, now I'm going to break my promise and guess the problem. It seems to me your concern is that the "bootstrap" work is *only* available via the initialization event. Surely, there's nothing preventing a visualizer from giving multiple paths towards invoking the bootstrap work. One via an initialization event (convenient for those simulating) and one through an explicit API (convenient for those who aren't).

EricCousineau-TRI, 2020-08-13:
I would like to just not care about initialization -- a visualizer in a diagram should be able to do sufficient bookkeeping to know when to publish on its own, without it really needing to be an event.

Perhaps once #10482 is resolved (e.g. `DrakeVisualizer` systems only consume `QueryObject` and can access geometry), then it'll all be moot.

> Both of these would fail if it needed to rebroadcast.

When would it need to rebroadcast?
Geometry change? If so, how do initialization events even a solution in the first place?

> Using a cache entry for behavior would break if caching were disabled (which is anti-caching).

Aye, gotcha.

> One via an initialization event (convenient for those simulating) and one through an explicit API (convenient for those who aren't).

Nah... That would make things even murkier, and requires me to carry around even more junk than before.

jwnimmer-tri, 2020-08-17:
For the purposes of this comment, I'll use "rendering binary" to refer to an out-of-process visualizer that receives configuration and update messages from a simulation process.  One example of a rendering binary would be `drake_visualizer`, though it applies equally well to other, similar programs.

Here is the key thesis:
1. If I craft a diagram that publishes information to a specific rendering binary (e.g., if I craft a diagram to send information on channel _foo_ at network address _bar_),
2. If the correctness of the published information is coupled to the remote state of the rendering binary (e.g., if a matching load message must always precede a draw message, or if draw messages are incremental instead of standalone, or ...),
3. Then either:
  a. The publisher must interrogate the rendering binary for its remote state before deciding what to send, or else
  b. The publisher must keep a local model of the remote state of the rendering binary, and
    1. The storage for the local model must have a congruent scope to the rendering binary -- nominally a process-global variable in the simulation to match the process-global state of the rendering binary itself; and
    2. In particular the local model _must not_ be stored in the context, because contexts are necessarily ephemeral within the simulation process and exist in greater quantity than the simulation itself.

One practical way to do this correctly with the (3b) approach is:
  - The simulation process has exactly one simulator, which has exactly one diagram, which has exactly one "publish to the rendering binary" leaf system.
  - The "publish to the rendering binary" leaf system has a mutable member field with its local model of the rendering binary, which it can consult to determine when additional information must be published.

For example with drake_visualizer, the mutable member field could be "What was the most recent viewer_load message I have transmitted", and to transmit a new load message iff the geometry data now differs.

To the question of "I rebooted the visualizer, so my publisher has to know to send load again" we could offer an api to reset the local model, or we could move to the (3a) approach of bi-directional information flow.

In case we do have multiple instance of a rendering binary (e.g., on different network addresses), we can have multiple publishers within our simulation process, as long as there the 1:1 relationship remains intact between publisher and rendering binary.

In any case, the main takeaways are that:

(1) Simulator::Initialize has _nothing at all_ to do with our publisher's local model of the rendering binary's state, so there is no reason to involve our rendering binary publishers with initialization events.

(2) Our local state that governs communication with _out of process entities_ must not live in the context -- we don't launch a rendering binary every time a context is allocated, so our model of its state must also not live within the context.

SeanCurtis-TRI, 2020-08-17:
@jwnimmer-tri Thanks for this comment. Very clear and very complete.

Clearly, to satisfy this issue in any reasonable way, we need to satisfy the prerequisite of a "local model" that can be easily analyzed.  This has been broached in the past and would be satisfied by a (or set of) geometry serial number(s) that would communicate how various aspects of `SceneGraph` have changed. At the coarsest level, a single `SceneGraph`-global serial number would be sufficient (but inefficient). *Technically*, that isn't strictly required -- a broadcasting `LeafSystem` could build its own view of `SceneGraph`'s world and compare against it at every publication, but, of course, no one would advocate that. (Implicit in this is the assumption that making sure that this system is the *only* broadcaster to the rendering binary would be the responsibility of the user.)

Also, worth emphasizing: the choice of *explicitly* adding mutable `LeafSystem` state in the *diagram*, undercutting the `const`-`non-const` paradigm ostensibly presented by the `System`-`Context` dichotomy.

jwnimmer-tri, 2020-08-17:
> (Implicit in this is the assumption that making sure that this system is the only broadcaster to the rendering binary would be the responsibility of the user.)

Exactly -- unless we do (3b) and interrogate it live.  At some point, I would rather we replace the viewer_load & viewer_draw messages with something less brittle as a protocol.

> Also, worth emphasizing: the choice of explicitly adding mutable LeafSystem state in the diagram, undercutting the const-non-const paradigm ostensibly presented by the System-Context dichotomy.

Exactly -- the point being that the rendering binary's remote status is not effectively `const` in any meaningful sense of the context.

jwnimmer-tri, 2020-10-29:
Note that the new #14247 does this correctly, with no initialization event.

EricCousineau-TRI, 2020-10-29:
^ FYI @sloretz regarding rviz prototype

jwnimmer-tri, 2020-12-09:
For the record, the only remaining offenders seem to be:
```
./bindings/pydrake/systems/meshcat_visualizer.py:235:        self.DeclareInitializationEvent(
./bindings/pydrake/systems/pyplot_visualizer.py:64:        self.DeclareInitializationEvent(
```

jwnimmer-tri, 2023-07-30:
The only remaining problematic code is the `bindings/pydrake/systems/pyplot_visualizer.py` system.