RobotLocomotion/drake #13779: Support tagged releases in setup_drake_colab.py

RussTedrake, 2020-07-31:
Currently, [setup_drake_colab](https://github.com/RobotLocomotion/drake/blob/e727721d7788945ef9e8a24c5a6f4d25c0d02afc/tools/install/colab/setup_drake_colab.py) supports three types of builds: `nightly`, `continuous`, and `experimental`.

I think we should offer `release` and have the version be the release number.  And using the most recent release should be the default.  That seems like the most durable thing we can/should offer?

jwnimmer-tri, 2020-07-31:
As of now, we only push the release binaries onto the [github release pages ](https://github.com/RobotLocomotion/drake/releases), and not into S3 like we do with the nightlies (there is no `https://drake-packages.csail.mit.edu/drake/tag` folder yet).  We _could_ teach the `setup_drake_colab` pull from github releases, but it's probably just worth pushing the tagged releases onto S3 and keeping everything in one place for colab.

Unlike the nightly, continuous, experimental, etc. folders on S3, we should set the tag folder to never expire.

> Using the most recent release should be the default.

Seems good.  But this is a property of the notebook cell, not a default in `setup_drake_colab`, I think?  Because the cell has to specify which version of the setup trampoline it fetches, it needs to pass the same version information when calling the trampoline.

RussTedrake, 2020-08-02:
I definitely agree that it must be done in the notebook cell.  I simply meant that we should set the `build` argument to default to `release` to help encourage good behavior.

jwnimmer-tri, 2020-08-02:
I'd actually argue for removing the default from the python script entirely (it's defaulting to `nightly` now).  Since the notebook cell already must specify the build type in order to fetch the trampoline, it's easy to pass the same type into the python function, and remove any chance of accidental version skew.

RussTedrake, 2020-08-03:
I'm fine with that, too.

RussTedrake, 2020-08-27:
Upgrading to priority high.  Starting next Tuesday, I will be distributing notebooks broadly with a ticking time bomb in the first cell (download script pointing to the nightly binaries).

RussTedrake, 2020-09-14:
Where do we stand on this?  Are the binaries from today's release available on drake-packages.csail?

jamiesnape, 2020-09-16:
Binaries are available, but it is a manual upload at present. I am working on it.

`https://drake-packages.csail.mit.edu/drake/releases/drake-X.Y.Z-{bionic|focal|mac|src}.tar.gz` where `X.Y.Z` ranges from 0.10.0 through 0.23.0.

The challenge is that there is nothing in the repo when it is tagged that says what the binary will be called. Searching for text in the changelog or using the GitHub API after the fact are the awkward solutions.

My ideal for many reasons would be having the changelog as the tagged commit (i.e., fork release branch, commit changelog, tag release, merge release branch to master, delete release branch).

jamiesnape, 2020-09-16:
I guess the date of the last commit is an indicator, but we sometimes have multiple builds per date, especially if there is a failure, so it is not authoritative.

jwnimmer-tri, 2021-01-21:
Lowering priority -- we have a manual solution for now.  As part of a bigger swing a reworking the release process we will work on automating this, but for the moment it's passable.

closed 2022-01-26 by RussTedrake