RobotLocomotion/drake #13780: Offer `setup_drake_colab.py` bundled with binary releases

RussTedrake, 2020-07-31:
The current proposed preamble for colab looks like this:
```
if 'google.colab' in sys.modules and importlib.util.find_spec('pydrake') is None:
  urlretrieve("https://raw.githubusercontent.com/RobotLocomotion/drake/master/tools/install/colab/setup_drake_colab.py",
              "setup_drake_colab.py")
  from setup_drake_colab import setup_drake
  setup_drake(version='20200730', build='nightly')
```

The request is that it looks, instead, something like this:
```
if 'google.colab' in sys.modules and importlib.util.find_spec('pydrake') is None:
  build = 'nightly'
  version = '20200730'
  urlretrieve(f"https://drake-packages.csail.mit.edu/drake/{build}/{version}/setup_drake_colab.py",
              "setup_drake_colab.py")
  from setup_drake_colab import setup_drake
  setup_drake(version=version, build=build)
```
That way the install script is locked to a revision and will remain durable, even if `setup_drake_colab.py` moves forward on `master`.

jwnimmer-tri, 2020-07-31:
Fully agree.

Another motivation is that we should not use `raw.githubusercontent...` URLs very much, because github has rate limiting for those and will temporarily reject fetching from there if over-used.

BetsyMcPhail, 2020-08-13:
CI PR [#114](https://github.com/RobotLocomotion/drake-ci/pull/114)

closed 2020-08-18 by jamiesnape

RussTedrake, 2020-08-27:
N.B. - I have updated the [setup script example](https://colab.research.google.com/drive/1c4A_Pvk_sLeLx0DpV0ak8dIaP2xT-ZO1) to use this.