RobotLocomotion/drake #13794: Documentation for `EventStatus` and `Simulator::set_monitor` led me astray

RussTedrake, 2020-08-02:
As documented in #13793, doxygen led me to believe that `Publish` events could return an `EventStatus` to terminate the simulation.  They cannot (yet).  

In particular, the documentation for `ReachedTermination` led me to a `set_monitor` example, which says "A monitor() function behaves the same as would a per-step Publish event handler included in the top-level System or Diagram being simulated."  I believe that is incorrect.  Please update it?

Should we have another issue to track the need to actually finish plumbing the `EventStatus`?

sherm1, 2020-08-03:
I guess #4447 is that issue but it is somewhat buried in the long discussion there. I'll open a clean one.

jwnimmer-tri, 2023-09-26:
BTW see #20035.  I think it will mostly (or fully?) solve this issue.

closed 2023-10-03 by sherm1