RobotLocomotion/drake #13797: `cmake --find-package ... -DNAME=drake` does not work?

EricCousineau-TRI, 2020-08-03:
When trying out different solutions for #7889, I wanted to see if I could get the include paths for Drake "for free" using CMake in `pkg-config` mode:
```sh
# Downloaded and unpackaged a nightly here.
$ cat ./share/doc/drake/VERSION.TXT
20200729064618 2daabfac1b81f9165d8fece7891df03bb67e8e72
$ cmake --version
cmake version 3.10.2
$ cmake --find-package -DCMAKE_PREFIX_PATH=${PWD} -DNAME=drake -DCOMPILER_ID=GNU -DLANGUAGE=CXX -DMODE=COMPILE
CMake Error in lib/cmake/drake/drake-config.cmake:
  cmake_policy PUSH without matching POP
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/CMakeFindPackageMode.cmake:180 (find_package)

drake not found.
CMake Error: Problem processing arguments. Aborting.
```

FTR, there *is* a correspond POP in the cps-generated `drake-config.cmake` file

EricCousineau-TRI, 2020-08-03:
Tried with a later version of CMake, same result:
```
$ fhs-extend ~/.local/opt/cmake-3.18.0-rc4-Linux-x86_64/
$ cmake --version
cmake version 3.18.0-rc4
$ cmake --find-package -DCMAKE_PREFIX_PATH=${PWD} -DNAME=drake -DCOMPILER_ID=GNU -DLANGUAGE=C -DMODE=COMPILE
CMake Error in lib/cmake/drake/drake-config.cmake:
  cmake_policy PUSH without matching POP
Call Stack (most recent call first):
  /home/eacousineau/.local/opt/cmake-3.18.0-rc4-Linux-x86_64/share/cmake-3.18/Modules/CMakeFindPackageMode.cmake:183 (find_package)
```

EricCousineau-TRI, 2020-08-03:
This failure mode doesn't seem relevant, but perhaps `pycps` can have an option to not put in `cmake_policy()`?
https://gitlab.kitware.com/cmake/cmake/-/issues/17612

jamiesnape, 2020-08-03:
`cmake --find-package` has probably always been broken, as it is on most CMake projects: https://cmake.org/cmake/help/latest/manual/cmake.1.html#run-the-find-package-tool. Not something we will fix.

EricCousineau-TRI, 2020-08-03:
That's kinda buried. If it's not well supported, I think y'all should hoist that message even higher.

For example, on this semi-bleeding edge cmake version:
```
$ cmake --version
cmake version 3.18.0-rc4
$ cmake --help | grep '\--find-package' -C 1
  -P <file>                    = Process script mode.
  --find-package               = Run in pkg-config like mode.
  --graphviz=[file]            = Generate graphviz of dependencies, see
```

Can you please file the appropriate upstream issue, and then post it here and re-close this issue once you've done that?

jamiesnape, 2020-08-04:
You are welcome to file the issue, but I think it is fine.

jwnimmer-tri, 2020-08-04:
This is not a Drake-specific issue, so I don't see any need to keep this ticket open on the `component: build system` dashboard, or spend Drake's dime on it.

closed 2020-08-04 by jwnimmer-tri

EricCousineau-TRI, 2020-08-04:
Filed: https://gitlab.kitware.com/cmake/cmake/-/issues/21056