RobotLocomotion/drake #13799: EventStatus plumbing needs to be hooked up

sherm1, 2020-08-03:
Currently event handlers can return an EventStatus that can be used to request termination or interruption of a simulation (as originally requested in #4447). However, except in the case of the Simulator's monitor() function, that status is ignored. That's because the event dispatcher virtuals aren't declared to return EventStatus and to do so would be a breaking change. It may be time to get that over with (see e.g. #13794 and #13793).

jwnimmer-tri, 2023-08-01:
> ... and to do so would be a breaking change ...

Usually this should be deprecate-able.  We introduce new virtuals with new names (with the new return type), and have one call the other.

sherm1, 2023-08-01:
The pure virtual dispatchers likely make that deprecation strategy awkward. However, looking this over I'm thinking the breakage could be minimal now that we have succeeded in (mostly) elimating user-overridden dispatchers in favor of real handler functions. The most commonly used APIs already accommodate handlers that either return EventStatus or void. I'll work up a PR and we can see how awful it is.

jwnimmer-tri, 2023-08-01:
(Assuming the deprecation hypothetical -- during the deprecation period both spellings of the virtual would be non-pure.)

jwnimmer-tri, 2023-09-26:
BTW see #20035.  I think it will mostly (or fully?) solve this issue.

closed 2023-10-03 by sherm1