RobotLocomotion/drake #13815: CLion instructions: add directions for debugging arbitrary test and example programs

rpoyner-tri, 2020-08-04:
This item arose from collaboration with @sherm1 on converting @mposa's Cassie benchmarks -- see #13782. It is not obvious how to enable debugging for arbitrary tests and programs within a Drake/CLion project. Apparently there is some auto-detection implemented somewhere for programs that contain the gtest.h header, but adding that to programs that don't need it is pretty mislading and generally sad.

Instead, I have written some instructions for enabling debugging of any program in the project. Victory condition for this ticket is to add those instructions to the Drake CLion documentation.

rpoyner-tri, 2020-08-04:
Doc draft so far:

```
.. -*- rst -*-
====================================================
 Debugging an arbitrary program in Drake with CLion
====================================================

Apparently CLion (or perhaps the Bazel plugin) has a certain amount of
auto-configuration of run/debug targets. It appears to hinge on the presence of
the gtest.h header in source files. This is convenient, but only further
mystifies the process of debugging a non-gtest program. This guide explains how
to configure debugging support for arbitrary programs in a Drake/CLion project.

This guide assumes all of the Drake-recommended installation and configuration
is done.

Get the bazel target string
===========================

Find the source file of the program in the tree view (left pane). Right-click
on the file, and select "Copy BUILD Target String". This will put the Bazel
target name into the clipboard.

Start a run configuration
=========================

From the top menu, select "Run/Edit Configurations...". Select the "+" at the
upper left of the dialog to add a new configuration. From the list, select
"Bazel Command".

Fill in the configuration
=========================

Now it's time to fill in the new blank configuration. Give it a name, then
select the "+" at the right side to add the target expression. Once the edit
box appears, paste the contents of the clipboard there. In "Bazel command",
select either "run" (for an arbitrary program), or "test" (for a Bazel test
target). Everything else can be left at default values. Click OK to finish.

Launch the debugger
===================

At this point, the top menu "Run" should have entries to run or debug your new
configuration. Select the debug entry there, or use the controls at the upper
right to launch the debugger.
```

rpoyner-tri, 2020-08-04:
I need to incorporate some early feedback from @sherm1, integrate the text into existing docs somewhere, and open a PR.

DamrongGuoy, 2020-08-04:
Such documentation would be very helpful. I report my experience that if I need to debug from drake into another library (e.g. FCL), I need to do these:
1. Go to `GDB` panel in CLion.
2. Enter gdb `dir` command to point to `bazel-drake` directory like this:
```
dir /home/damrongguoy/GitHub/DamrongGuoy/drake/bazel-drake
```
[image]

Otherwise, I got assembly code instead of source code when I stepped from drake into FCL.

sherm1, 2020-08-04:
@DamrongGuoy nice -- thanks! How do you get CLion to give you (gdb) command line access?

DamrongGuoy, 2020-08-05:
The `GDB` panel is next to the `Variables` panel in `Debug` window. I highlight it with red mark in this picture:

[image]

PR #13840 (2020-08-10): CLion doc: Add section on debugging arbitrary programs
Resolves #13815.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13840)
commit 5fe2dbec (2020-08-10): CLion doc: Add section on debugging arbitrary programs (#13840)

* CLion doc: Add section on debugging arbitrary programs

Resolves #13815.
closed 2020-08-10 by SeanCurtis-TRI