RobotLocomotion/drake #13818: High count collision geometry causes segfault

chenheamazon, 2020-08-04:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
Yes
> * Have you reviewed existing GitHub issues and StackOverflow posts?
Yes the existing segfault issues are not related to this current one.
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?
Yes

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

This is on commit 79200833ded29b88f01a28b74afacdd9d0323e87 from June 1, 2020

When I try to run the simple_gripper built in example, and change the number of collision geometry in the example https://github.com/RobotLocomotion/drake/blob/master/examples/simple_gripper/simple_gripper.cc#L67 (this line here) from 8 to 300, it causes a segfault. We have a similar use case where too many registered collision geometry causes a segfault and we suspect that this is the same issue.

sherm1, 2020-08-04:
Assigning to @SeanCurtis-TRI for initial disposition.

SeanCurtis-TRI, 2020-08-04:
I suspect that this got resolved with #13676.

SeanCurtis-TRI, 2020-08-04:
You should see the benefit in the next release (or in master). Based on what the issue says, you should be able to reduce the number of collision geometries to around 100 (to limit the number of contacts to <= 128) and be ok. Once you have access to master (or the next release), you won't have to live with that artificial constraint.

jwnimmer-tri, 2020-08-04:
FYI #13676 is available as of v0.20.0 (and v0.21.0, etc.) tags.

SeanCurtis-TRI, 2020-08-04:
I'm surprised; I went looking in the [release notes](https://drake.mit.edu/release_notes/v0.21.0.html) and didn't see it. But I hadn't actually dug in the commit history to confirm it was included.

jwnimmer-tri, 2020-08-04:
https://drake.mit.edu/release_notes/v0.20.0.html#multibody-dynamics

SeanCurtis-TRI, 2020-08-04:
Oops...because it's [here](https://drake.mit.edu/release_notes/v0.20.0.html).

SeanCurtis-TRI, 2021-06-29:
@chenheamazon Did you ever have the chance to confirm that the problem has gone away?

SeanCurtis-TRI, 2021-12-04:
Closing because there's been no action for 1.5 years.

closed 2021-12-04 by SeanCurtis-TRI